maximkulkin/esp-homekit-demo #303: Security Problem

okanduzyel, 2020-01-23:
Hi,

There is an enormous problem with examples especially sonoff examples.

Please consider the case that to cut the electricity of home and connect it again. 

Modem boot up time takes around ~1 minute. At this time all sonoff devices, or a device which has wifi configuration page like sonoff, publish their wifi configuration page. And now, they can be handled by someone. 

A neighbor has your diy examples at home. To cut the neighbor's electricity and give it again. Wait 5-10 seconds and start handle them with your iPhone.

**Suggestion:** There is a reset configuration button at every sonoff examples. Without pressing this button, the configuration page mustn't open.

lrusnac, 2020-01-24:
I don't think that's a good suggestion. 
the solution is to add a password for the configuration AP instead, and have the esp try to connect to the known AP every X minutes if not connected already

okanduzyel, 2020-01-24:
Hacker doesn't need to add accessories to its home. Please think about that if hacked added accessories its home, what it would do? Open and close the lamps remotely? I think Hacker's aim is breaking the actual system. Can He or she do it? The answer is yes. Someone can do it. So, for me, this system is not reliable at that point. 

Here is the second question. What does the configuration page actually? First time of setting up the accessory for home OR If users want to change SSID of home. Briefly, In both case, user can press the physical configuration button and set accessory up quickly.

peros550, 2020-01-24:
@okanduzyel 

Your solution applies to a subset of esp8266 devices, those that have physical button (Sonoff, etc). For devices that lack physical button what would you propose ?

maximkulkin, 2020-01-25:
Also, there are devices that are hard to reach.

What's wrong with protecting your wifi_config AP with a password? If someone can hack your wifi_config WiFi password, what will stop him from hacking your real WiFi password?!

Second, indeed, wifi_config configuration page does not allow much. Worst case somebody will make your device unavailable by connecting it to a different WiFi network. If device was already paired, hacker can not pair again and control the device, he will first need to reset it's state, which is not possible remotely.

okanduzyel, 2020-01-25:
Okay guys.

Could you explain for me why it is appearing when router offline or booting?  I can't find a reason for that. 

I cut my home's electricity in order to show you want happens. 

[url=https://hizliresim.com/anV4qQ][img]https://i.hizliresim.com/anV4qQ.jpg[/img][/url]

My neighbors can see them can can change their wifi SSID. My question is why?

I am at work and router has a problem. All day they will be visible. My curious neighbors or bad guys can deal with them. Why? If my router has a problem, It can try to always connect my router. Why It also publish that configuration page?

-Some devices can not reachable?
Why should I change their SSID?

-One more pin required?
ESP12E has many? (pin0 is the best one for that job)

maximkulkin, 2020-01-25:
> My neighbors can see them and can change their wifi SSID. My question is simple. Why?

Because you do not bother setting up WiFi passwords on your accessories. In my private firmwares all wifi-config APs are password protected.

lrusnac, 2020-01-26:
alternatively can just hardcode the wifi credentials in your sketch and not use the wificonfig library..

okanduzyel, 2020-01-28:
I am going to try explaining the security issue again. 

I had cut my home's electricity(with main switch at outside) after 15 seconds, gave it again. Then, My neighbor opened him iPhone's wifi menu and saw entire sonoff devices which were at my home. 

He opened each configuration page of sonoff devices and changed their SSID quickly. Meantime, I was thinking, electric gone and come, why my devices looks "no response"? unplug-plug and there is no difference. What happens here? Tried everything but you know there is only way that reset by hand, open its configuration page and add accessory to my home app.

After a short time, my neighbor did it again and again. Maybe tomorrow when I am at job, maybe I was in supermarket etc. 

My question is very simple. WHY is there a configuration page? I don't wanna reset my devices' SSID configuration even my router has a problem or the router is not found by sonoff devices. 

I hope, this is clear for you to understand what I exactly mean.

maximkulkin, 2020-01-28:
And I will repeat it again: when you initialize wifi_config, pass non-empty password (2nd argument) and your neighbor won’t be able to connect to your devices and change their network.

RavenSystem, 2020-01-28:
@okanduzyel This repo contains code examples of how to use esp-homekit library to help developers understanding its functions and way to work. It is not focused on create final versions of firmwares to be installed by users.

closed 2020-01-30 by okanduzyel