maximkulkin/esp-homekit-demo #307: haw sanyone got this working on an esp-8285?

mrTuomoK, 2020-02-06:
I got the led example working on a esp-8266 but unable to get it working on a esp-8285. Everything seems to go fine and i've tried different flashing params but still no good... got some csum error in serial monitor after flashing.

mrTuomoK, 2020-02-06:
Ok now I was aable to flash it almost correctly by using dout -mode ... now it's failing to write the accessory key etc. to it's memory... any ideas with what mem address params I should use to flash it with? (or is the problem something else?)

maximkulkin, 2020-02-06:
0x7A000 is a safe bet

mrTuomoK, 2020-02-06:
it seems that the "make -C examples/led flash" command still writes to wrong mem address even though I have used export HOMEKIT_SPI_FALSH_BASE_ADDR=0x7a000=

esptool.py v2.8
Serial port /dev/tty.wchusbserial1440
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 60:01:94:9b:c5:90
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0020
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 114.3 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 585.6 kbit/s)...
Hash of data verified.
Compressed 432544 bytes to 311120...
Wrote 432544 bytes (311120 compressed) at 0x00002000 in 27.9 seconds (effective 124.2 kbit/s)...
Hash of data verified.

mrTuomoK, 2020-02-06:
With this command-set I'm the closest to get it done and working but get the failing to save 

esptool.py -p /dev/tty.wchusbserial1440 --baud 57600 write_flash -fs 1MB -fm dout -ff 26m 0x0 xxx/homekit_esp/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin &&
esptool.py -p /dev/tty.wchusbserial1440 --baud 57600 write_flash -fs 1MB -fm dout -ff 26m 0x1000 xxx/homekit_esp/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin &&
esptool.py -p /dev/tty.wchusbserial1440 --baud 57600 write_flash -fs 1MB -fm dout -ff 26m 0x2000vxxx/homekit_esp/esp-homekit-demo/examples/led/firmware/led.bin

SO basically I'm flashing the parts one by one so is some of them wrong? Tried flashing the led.bin to 0x7a00 with less success

mrTuomoK, 2020-02-06:
I have also tried to rebuild the app, clearing the flash and so on...

maximkulkin, 2020-02-06:
If you're building it inside docker container like I do, environment variables are not automatically propagated into container. You have several options: pass environment variable to docker explicitly (with `-e MY_ENV_VAR`)
```shell
export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7A000
docker run -it --rm -v `pwd`:/project -w /project esp-rtos -e HOMEKIT_SPI_FLASH_BASE_ADDR make -C examples/led all
```
or pass it as make argument:
```shell
docker run -it --rm -v `pwd`:/project -w /project esp-rtos make -C examples/led HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000 all
```
or just edit examples/led/Makefile and add this line before `include $(SDK_PATH)/common.mk`:
```Makefile
HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000
```

mrTuomoK, 2020-02-06:
Thanks @maximkulkin ! Just figured it out myself too and seems to work :D

closed 2020-02-06 by maximkulkin