maximkulkin/esp-homekit-demo #308: Strange problem son off mini using the examples switch

kenthinson, 2020-02-10:
I compiled and loaded several son off mini with the example switch project. They have been working well for a few months. Today they all disconnected from my network. Then they all went into wifi AP mode. If I look at my wifi on my phone or computer I see a bunch of sonoff-switch-blahblahblah. It's very strange I did not make any change to my network. They all deauthenticated at the same time and refuse to work now. I tried power cycling them. They will show as online and join the network like normal for about 5 to 10 seconds. Then they disconnect and go back to sonoff-switch-blahblahblah. How should I go about troubleshooting this?

maximkulkin, 2020-02-13:
Sun flares causing electromagnetic disturbances?

kenthinson, 2020-04-30:
Yeah it was real weird. They all started working on there own a few hours later. I’m closing this issue. Sorry I had forgotten I had opened it.

closed 2020-04-30 by kenthinson