maximkulkin/esp-homekit-demo #309: Failed to resolve component 'http-parser'

timmytjieme, 2020-02-12:
Good day, I'm using esp-idf to program my esp32 I want to use the esp32 led example but got the following error on my cmd: 

```
C:\Users\Timmy\Desktop\esp\esp-homekit-demo\examples\esp32\led>idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\Users\Timmy\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\timmy\desktop\esp\esp-homekit-demo\examples\esp32\led\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\timmy\desktop\esp\esp-homekit-demo\examples\esp32\led"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 91209bd
-- Building ESP-IDF components for target esp32
CMake Error at C:/Users/Timmy/Desktop/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'http-parser'.
Call Stack (most recent call first):
  C:/Users/Timmy/Desktop/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  C:/Users/Timmy/Desktop/esp-idf/tools/cmake/build.cmake:212 (__build_expand_requirements)
  C:/Users/Timmy/Desktop/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  C:/Users/Timmy/Desktop/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Timmy/Desktop/esp/esp-homekit-demo/examples/esp32/led/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

and my output log:

```
The target system is: Generic -  - 
The host system is: Windows - 10.0.18362 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "C:/Users/Timmy/Desktop/esp/esp-homekit-demo/examples/esp32/led/build/CMakeFiles/3.13.4/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: C:/Users/Timmy/espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
0

[... 541 lines cut ...]
    Feature record: CXX_FEATURE:0cxx_range_for
    Feature record: CXX_FEATURE:0cxx_raw_string_literals
    Feature record: CXX_FEATURE:0cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:0cxx_relaxed_constexpr
    Feature record: CXX_FEATURE:0cxx_return_type_deduction
    Feature record: CXX_FEATURE:0cxx_right_angle_brackets
    Feature record: CXX_FEATURE:0cxx_rvalue_references
    Feature record: CXX_FEATURE:0cxx_sizeof_member
    Feature record: CXX_FEATURE:0cxx_static_assert
    Feature record: CXX_FEATURE:0cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates

```

who can this be solved? I've also used
 `git submodule update --init --recursive`

maximkulkin, 2020-02-12:
`make all`

timmytjieme, 2020-02-12:
> make all

Tried `make all` But receive an error 

`‘make’ is not recognized as an internal or external command, operable program or batch file.`

maximkulkin, 2020-02-12:
You need `make` to build, go get it

timmytjieme, 2020-02-12:
> You need `make` to build, go get it

I’ve used the esp-idf-tools-setup-2.2.exe with 
Python 3.7
Git 2.21.0
Esp-idf v3.3.1

Is this correct or do i need additional software?

maximkulkin, 2020-02-12:
Sorry, I do not use Windows and do not do Windows support.

timmytjieme, 2020-02-13:
> Sorry, I do not use Windows and do not do Windows support.

Would you recommend i use linux like ubuntu?

maximkulkin, 2020-02-13:
Yes, Ubuntu is da best

timmytjieme, 2020-02-13:
Did enjoy programing within rasbian on my raspberry pi think I’ll allocate one of my drives for Ubuntu will give it a try to program my microcontroller. Thanks for the help, much appreciated.

pocketscience, 2020-03-04:
Did you include extras/http-parser in your makefile??

FWIW, get a Mac... da best! :)

mriksman, 2020-03-12:
Did you have http-parser in your components folder?
```
project
 -main
     -led.c
  -components
     -homekit
     -wolfssl
     -http-parser
```

closed 2020-06-26 by maximkulkin

ykutovoy, 2022-01-28:
Hi to all
I understand that it was far ago, but 
how to resolve this issue?
I try to build project on ubuntu and get the same error

`CMake Error at /esp/esp-idf/tools/cmake/build.cmake:197 (message):
  Failed to resolve component 'nghttp'.
`

Could you say what to correct