maximkulkin/esp-homekit-demo #310: Problem with button example

renandw, 2020-02-17:
Hello,

I managed to flash successfully the Button example. It was working fine with all 3 types of pressing configuration. But the batteries died and when I changed those, the button was not working.
I erased the ESP and flashed again. And it works fine. I tried to power off an on the ESP to repeat the power outage. And it did not work as the first time.
Then I ran the "make test" command to see details.

`ESP-Open-SDK ver: 0.0.1 compiled @ Jan 28 2020 22:53:27
phy ver: 273, pp ver: 8.3

mode : sta(2c:f4:32:12:91:5a)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 24:5E:EB:81:F9:7E
>>> HomeKit: Found admin pairing with FF8F5239-0D1E-401F-9801-D108619DFA94, disabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Botão Inteligente 	md=Botãopv=1.0id=24:5E:EB:81:F9:7Ec#=1s#=1ff=0sf=0ci=15
                       sh=19txeg== Port=5556 TTL=4500
scandone
add 0
aid 2
cnt

connected with Mansao SP, channel 1
dhcp client start...
ip:192.168.15.27,mask:255.255.255.0,gw:192.168.15.1
single press
single press
double press
single press
single press
single press
double press
long press
long press
long press
long press
long press
long press
single press
long press
double press
single press
single press
double press`

I noticed that the button is working and the ESP detects the presses. But the accessories configured was not affected to the presses.
I tried reseting the ESP sometimes with inconsistent results. Sometime it right away works, sometimes after sometime. and sometime nothing at all.

`ESP-Open-SDK ver: 0.0.1 compiled @ Jan 28 2020 22:53:27
phy ver: 273, pp ver: 8.3

mode : sta(2c:f4:32:12:91:5a)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 24:5E:EB:81:F9:7E
>>> HomeKit: Found admin pairing with FF8F5239-0D1E-401F-9801-D108619DFA94, disabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Botão Inteligente 	md=Botãopv=1.0id=24:5E:EB:81:F9:7Ec#=1s#=1ff=0sf=0ci=15
                       sh=19txeg== Port=5556 TTL=4500
scandone
add 0
aid 2
cnt

connected with Mansao SP, channel 1
dhcp client start...
ip:192.168.15.27,mask:255.255.255.0,gw:192.168.15.1
single press
single press
double press
single press
single press
single press
double press
long press
long press
double press
>>> HomeKit: Got new client connection: 4 from 192.168.15.19
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with FF8F523E-401F-9801-D108619DFA94
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories`

Anybody going through this?
Thanks

renandw, 2020-02-17:
Ps: I have an Apple TV 4 HD in the house as central

renandw, 2020-02-21:
I understand that this button example needs a home central, like Apple TV, HomePod, or an iPad. It is possible that I need to reconfigure the central (in my case, the Apple TV?
When I pair it works just fine, but when reboot it doesn’t even show the message “no response”.

maximkulkin, 2020-02-24:
I'm confused on what's happening. Does it briefly show "Updating" when you re-open your Home.app ?
What are those logs for ? Do those clicks happen spontaneously or are they just result of user interaction?
Does your device show up in same network your Apple TV is at? Does it broadcast notification about itself through mDNS ?

renandw, 2020-02-26:
I flash the button example, and configure it normally. It is responsive to one press, double presses or long press. All works really well.
But when I power it off and power it on, I get inconsistent results. Sometimes it works right away. Sometimes not at all.
When I close the home.app and reopen it. The others accessories show “updating”; but not the button accessory.
Then I ran “make test” command to see if the esp wasn’t recognizing the button pushes. But it show in “printf” (single press; double presses or long press).
I restored the Apple TV to factory settings and also my iPhone. Flashed the esp with “erase_flash first”. And paired it with HomeKit again in the same WiFi network.
[image]
[image]

renandw, 2020-02-26:
The presses are reflect of user interaction.
The pictures above show when I disconnect the accessories from power.
When I power the button on, the logs show the HomeKit starting server, connecting to WiFi, broadcasting mDNS…
But the button just works when I get this:

HomeKit: Got new client connection: 4 from 192.168.15.19
HomeKit: [Client 4] Pair Verify Step 1/2
HomeKit: [Client 4] Pair Verify Step 2/2
HomeKit: [Client 4] Found pairing with FF8F523E-401F-9801-D108619DFA94
HomeKit: [Client 4] Verification successful, secure session established
HomeKit: [Client 4] Get Accessories

maximkulkin, 2020-02-26:
I've built repro and I confirm that the problem exists: Apple TV does not reconnect after accessory power cycle.

renandw, 2020-02-27:
:(
thanks very much

sanilsdevan, 2020-04-10:
I too am facing the same issue.:(

 any workaround to this issue?

mine gets stuck at this once I power down and then turn on the button. if I click, it is able to recognize the clicks but no response on my led on another node mcu

md=MyButtonpv=1.0id=B0:C1:6B:4A:17:7Fc#=1s#=1ff=0sf=0ci=15sh=8nkrig== Port=5556 TTL=4500
single press
double press
long press

But once I open the home app on my Ipad & go to the button settings and TEST any of the options, ie single double or long-press, then it gets connected. From then on if I press the button it works.

`>>> HomeKit: Got new client connection: 4 from 192.168.0.169
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 7127ECD5-CA9A-4137-889A-48EC877333C7
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics

Also, if I don't open the home app on Ipad for nearly 15 to 20 mins, then OCASSIONALLY, it automatically completes the remaining steps gets connected. From then on, if I press the button, everything works fine.

sanilsdevan, 2020-04-11:
An update.

I kept the serial monitor on for the whole day. I find this error repeating approx 17 to 20 mins apart.

`
>>> HomeKit: [Client 5] Get Characteristics
 !!! HomeKit: [Client 4] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 4] Closing client connection
!!! HomeKit: [Client 5] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 5] Closing client connection
>>> HomeKit: Got new client connection: 4 from 192.168.0.169
`

This error repeats every 20 mins apart.

konifer44, 2020-05-28:
Any solution? Same problem here with HomePod

renandw, 2020-09-30:
I’ve been testing it in ios14 and tvos14. It is working just fine.

closed 2020-09-30 by renandw