maximkulkin/esp-homekit-demo #311: Cmake build idf v4.0

ginodecock, 2020-02-17:
Hi,

I failed to build the esp32 led sample using Cmake under idf v4.0.

Should this work?

maximkulkin, 2020-02-17:
No, use make

andalera, 2020-02-22:
I got it working on stock ESP-IDF 4.0 with CMake.
Here's my notes...  worked for me, ymmv...

Installed ESP-IDF per 4.0 getting started guide (Windows)
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html
I was also using the new Eclipse plugin, setup per guide:
https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md

--------------esp-homekit testing-------------------
Cloned esp-homekit-demo to a folder (not associated with any local project)
Created new EspressIf IDF project in Eclipse. named it HomeKit-Testing
Created components folder in my HomeKit-Testing project

Examined required dependencies of in CMakeLists.txt in esp-homekit-demo/components/homekit:
	it needs: wolfssl cJSON http-parser mdns
	Where to get them? The esp-homekit-demo source has everything not already in ESP-IDF:
	

- esp-homekit-demo/components/common/wolfssl -> has CMakeLists.txt
- 	esp-homekit-demo/components/esp8266-open-rtos/cJson -> no CMakeLists.txt
- 	esp-homekit-demo/esp-idf/http-parser -> has CMakeLists.txt
- 	mdns -> part of esp32 

		
I copied all these from the esp-homekit-demo locations to my project's components folder:
```
HomeKit-Testing
	/components
		cJson -> doesn't have CMakeLists.txt -> made one.
		homekit
		http-parser
		wolfssl
```

For the cJson component, I created a CMakeLists.txt file at components/cJSON and put this in it:
```
idf_component_register(
    SRC_DIRS "cJSON"
    INCLUDE_DIRS "cJSON"
)

```
For the homekit component, build errors were happening in port.h on the #include "esp_spi_flash.h"
I had to modify the CMakeLists.txt in homekit to add spi_flash as a required dependency (from ESP-IDF).
	
Once that modification was done, it built with no errors. (Lots of warnings though...)

Used the main/led.c from esp-homekit-demo/examples/esp32 for initial test.  Pasted the contents into main.c in my project.

- Commented out #include "wifi.h" - not needed and I don't have arduino-esp32 as a component on this project. 
- Added #include <stdbool.h> because eclipse was not happy about bools without it.
- Modified wifi_config area to use CONFIG_ESP_WIFI_SSID and CONFIG_ESP_WIFI_PASSWORD from the basic Eclipse IDF project
  - Kconfig.projbuild has an addition for ssid and pass to be set in sdkconfig.

edited homekit_server_config to have setupId per the main esp-homekit github (using gen_qrcode)
also changed the passcode to match the esp-homekit readme.md example.
```
homekit_server_config_t config = {
    .accessories = accessories,
    .password = "123-45-678",
	.setupId = "1QJ8"
};
```

`idf.py menuconfig`
set my ssid and pass in Example Configuration.

`idf.py build`
built successfully!

To generate QR code per the esp-homekit guidance will need qrcode installed in python.
https://pypi.org/project/qrcode/
`python -m pip install qrcode[pil]`
it should pull in Pillow as well for image processing. https://pillow.readthedocs.io/en/stable/installation.html

Note that the qrcode generator doesn't like 111-11-111 as the password.  I used 123-45-678 instead like the esp-homekit readme.md says.

From components\homekit\tools>
`python gen_qrcode 5 123-45-678 1QJ8 qrcode.png`
qrcode.png should now be setup.

Now that we know we can 1) build successfully and 2) generate the qrcode; lets get this flashed.
```
idf.py -p COM4 erase_flash
idf.py -p COM4 flash
idf.py -p COM4 monitor
```

No issues visible, all seems fine.

Setup with my phone using qrcode.png, and all is well! I can turn on and off the led on the esp32 just fine from Apple Home.

ginodecock, 2020-02-23:
I followed your manual and it works great. Thanks.

closed 2020-02-23 by ginodecock