maximkulkin/esp-homekit-demo #312: Update on wake up QUESTION

kiralikbeyin, 2020-02-26:
@maximkulkin 
I am thinking about to make a door sensor. 

ESP8266 will be in sleep when i run the HomeKit app and normally it will not get update.

Is there any trick to bypass or send a fake update?

maximkulkin, 2020-02-26:
Not that I am aware of. If controller does not connect to accessory (and connection is supposed to be initiated by controller), there is no way to let controllers know. Best bet is to use bridge devices that do not sleep and are always connected.

kiralikbeyin, 2020-02-26:
@maximkulkin Thanks. 
Maybe it is possible with a Bluetooth thing.. 

As you said using a bridge is exact solution.

closed 2020-02-26 by kiralikbeyin

Kristian8606, 2020-02-26:
i started building battery sensors, i have not finished the project yet but it is working.  the concept is the following door sensor consists of a small Arduino with a LoRa module that uses a 3v battery and sends data to esp with a LoRa module.  esp is configured as a bridge and is permanently plugged in.  In this way, the esp can accept multiple sensors and motion sensors.