maximkulkin/esp-homekit-demo #314: Changes  to be made to the button program to use esp-wifi-config library

sanilsdevan, 2020-03-01:
What changes should i make (what all to be added and also what all lines to be removed) in the button program so that i can use esp-wifi-config for configuring wifi access point on initial boot. I have tried many times but every time one or the other error turns up

pocketscience, 2020-03-04:
Have you looked at the wifi_config example..?

sanilsdevan, 2020-03-04:
[image]

how should I change the highlighted portion of the code in the button program

pocketscience, 2020-03-04:
It's pretty obvious isn't it...

void on_wifi_ready() {
    homekit_server_init(&config);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_config_init("my button", NULL, on_wifi_ready);
    if (button_create(BUTTON_PIN, button_callback)) {
        printf("Failed to initialize button\n");
    }
}

I haven't tested this but that should get you going...

sanilsdevan, 2020-03-05:
Did that .. all was fine with compiling. when it came to the erase flash command it gave the error
`make: Entering directory '/home/user/esp/esp-homekit-demo/examples/button'`
`make: *** No rule to make target 'erase'.  Stop.`
`make: Leaving directory '/home/user/esp/esp-homekit-demo/examples/button'
`
I used the command
`make -C examples/button erase flash`
in the esp-homekit-demo directory

pocketscience, 2020-03-05:
I believe you're missing an underscore...

maximkulkin, 2020-03-05:
It’s `erase_flash`, not `erase`

sanilsdevan, 2020-03-05:
ooo.. my my... thank you
everything worked ok
just that I am not able to get the double click working.. is there any particular timing between clicks.. will I be able to configure the click interval

I configured another esp8266 to respond to clicks
a single click turns on chip led
double click is supposed to turn it off
and long click to turn on the led for 1 minute and then turn off

single & long click works perfectly. but I am not able to get the double click working
I tried rapid clicking as well as 2 clicks with an interval of up to 1-2 secs.. but can't get it working

renandw, 2020-03-05:
I had this problem.
See this: https://github.com/maximkulkin/esp-button
When updated the code it works flawlessly

sanilsdevan, 2020-03-06:
oh..k

so should i be pulling a copy `esp-button` folder to `esp-homekit-demo/examples` 
and then move to `/esp-button/examples`

and compile the `button_example.c` using `Makefile`

what about 
`
homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "Sample LED");

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
            HOMEKIT_CHARACTERISTIC(
                ON, false,
                .getter=led_on_get,
                .setter=led_on_set
            ),
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};
`
& other wifi hotspot initialisation portion

renandw, 2020-03-06:
no.
in button example, you should change the current library "button.c" and "button.h" to the new one in "eps-button"

your code should look like this:

`#include <stdio.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>
#include "button.h"

#define BUTTON_PIN 0
#ifndef BUTTON_PIN
#error BUTTON_PIN is not specified
#endif

void button_identify(homekit_value_t _value) {
    printf("Button identify\n");
}

homekit_characteristic_t button_event = HOMEKIT_CHARACTERISTIC_(PROGRAMMABLE_SWITCH_EVENT, 0);

void button_callback(button_event_t event, void* context) {
    switch (event) {
        case button_event_single_press:
            printf("single press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(0));
            break;
        case button_event_double_press:
            printf("double press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(1));
            break;
        case button_event_long_press:
            printf("long press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(2));
            break;
        default:
            printf("unknown button event: %d\n", event);
    }
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(
        .id=1,
        .category=homekit_accessory_category_programmable_switch,
        .services=(homekit_service_t*[]) {
            HOMEKIT_SERVICE(
                ACCESSORY_INFORMATION,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                    HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
                    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0012345"),
                    HOMEKIT_CHARACTERISTIC(MODEL, "MyButton"),
                    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
                    HOMEKIT_CHARACTERISTIC(IDENTIFY, button_identify),
                    NULL
                },
            ),
            HOMEKIT_SERVICE(
                STATELESS_PROGRAMMABLE_SWITCH,
                .primary=true,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                    &button_event,
                    NULL
                },
            ),
            NULL
        },
    ),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void on_wifi_ready() {
        homekit_server_init(&config);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    button_config_t config = BUTTON_CONFIG(
        button_active_low,
        .long_press_time = 1000,
        .max_repeat_presses = 3,
    );

    wifi_config_init("botao", "serasevai", on_wifi_ready);

    int r = button_create(BUTTON_PIN, config, button_callback, NULL);
    if (r) {
        printf("Failed to initialize a button\n");
    }

}`

renandw, 2020-03-06:
replace the "button.c" and "button.h" from the current folder in button example to those in "eps-button".

and try this code below in main.c

```
#include <stdio.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>
#include "button.h"

#define BUTTON_PIN 0
#ifndef BUTTON_PIN
#error BUTTON_PIN is not specified
#endif

void button_identify(homekit_value_t _value) {
    printf("Button identify\n");
}

homekit_characteristic_t button_event = HOMEKIT_CHARACTERISTIC_(PROGRAMMABLE_SWITCH_EVENT, 0);

void button_callback(button_event_t event, void* context) {
    switch (event) {
        case button_event_single_press:
            printf("single press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(0));
            break;
        case button_event_double_press:
            printf("double press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(1));
            break;
        case button_event_long_press:
            printf("long press\n");
            homekit_characteristic_notify(&button_event, HOMEKIT_UINT8(2));
            break;
        default:
            printf("unknown button event: %d\n", event);
    }
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(
        .id=1,
        .category=homekit_accessory_category_programmable_switch,
        .services=(homekit_service_t*[]) {
            HOMEKIT_SERVICE(
                ACCESSORY_INFORMATION,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                    HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
                    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0012345"),
                    HOMEKIT_CHARACTERISTIC(MODEL, "MyButton"),
                    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
                    HOMEKIT_CHARACTERISTIC(IDENTIFY, button_identify),
                    NULL
                },
            ),
            HOMEKIT_SERVICE(
                STATELESS_PROGRAMMABLE_SWITCH,
                .primary=true,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                    &button_event,
                    NULL
                },
            ),
            NULL
        },
    ),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void on_wifi_ready() {
        homekit_server_init(&config);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    button_config_t config = BUTTON_CONFIG(
        button_active_low,
        .long_press_time = 1000,
        .max_repeat_presses = 3,
    );

    wifi_config_init("botao", "serasevai", on_wifi_ready);

    int r = button_create(BUTTON_PIN, config, button_callback, NULL);
    if (r) {
        printf("Failed to initialize a button\n");
    }

}
```

sanilsdevan, 2020-03-09:
Thank you so much.. Everything works perfectly now... Nd hey.. I took a copy of the those files from your github page.. @renandw

closed 2020-03-09 by sanilsdevan