maximkulkin/esp-homekit-demo #316: error xxx making C -examples

Jacquesduval, 2020-03-03:
Sorry to say got an error when i create the sample see copy of my terminal here below 

jacques@artdujardin esp-homekit-demo % sudo docker run -it --rm -v "$(pwd)":/project -w /project esp-rtos make -C examples/garage         
Password:
make: Entering directory '/project/examples/garage'
CC /opt/esp-open-rtos/open_esplibs/libmain/user_interface.c
mkdir: cannot create directory 'build': Permission denied
/opt/esp-open-rtos/open_esplibs/component.mk:15: recipe for target 'build/open_esplibs_libmain//user_interface.o' failed
make: *** [build/open_esplibs_libmain//user_interface.o] Error 1
make: Leaving directory '/project/examples/garage'
jacques@artdujardin esp-homekit-demo % 

Would you please help me ?
Thanks
Best Rgds

pocketscience, 2020-03-04:
It says it right there - you have a permissions issue with the /examples/garage folder. Check/fix the folder permissions. Not sure why you're using sudo either to run the build - that shouldn't be necessary.

Jacquesduval, 2020-03-04:
Thanks a lot for your answer, will have  a look and let you know.
How could i contribute to you ? I am new on Github and don’t know how to contribute ?
Best Rgds

> 
> It says it right there - you have a permissions issue with the /examples/garage folder. Check/fix the folder permissions. Not sure why you're using sudo either to run the build - that shouldn't be necessary.

Jacques Duval
3515 A route de Bonnieux
84560 Menerbes

Mail : jacquesduv@me.com
Phone : +33 607 019 661

pocketscience, 2020-03-04:
This isn't my repository - I'm using it just like you. One suggestion regarding contributions would be to learn as much as possible and try to help others who ask questions.

closed 2020-04-07 by maximkulkin