maximkulkin/esp-homekit-demo #317: Problems with the last code

MizzyAntunes, 2020-03-03:
vagrant@vagrant-ubuntu-trusty-64:~/esp-homekit-demo$ make -C examples/led test
make: Entering directory `/home/vagrant/esp-homekit-demo/examples/led'
CC /home/vagrant/esp-open-rtos/open_esplibs/libmain/ets_timer.c
make: xtensa-lx106-elf-gcc: Command not found
make: *** [build/open_esplibs_libmain//ets_timer.o] Error 127
make: Leaving directory `/home/vagrant/esp-homekit-demo/examples/led'
vagrant@vagrant-ubuntu-trusty-64:~/esp-homekit-demo$

Please, can somebody help me?

MizzyAntunes, 2020-03-03:
vagrant@vagrant-ubuntu-trusty-64:~$ cd esp-homekit-demo
vagrant@vagrant-ubuntu-trusty-64:~/esp-homekit-demo$ make -C examples/led test
make: Entering directory `/home/vagrant/esp-homekit-demo/examples/led'
CC /home/vagrant/esp-open-rtos/open_esplibs/libmain/ets_timer.c
make: xtensa-lx106-elf-gcc: Command not found
make: *** [build/open_esplibs_libmain//ets_timer.o] Error 127
make: Leaving directory `/home/vagrant/esp-homekit-demo/examples/led'
vagrant@vagrant-ubuntu-trusty-64:~/esp-homekit-demo$

pocketscience, 2020-03-04:
Did you follow the build instructions completely?

https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266

Did you have any errors during that?

MizzyAntunes, 2020-03-04:
> Did you follow the build instructions completely?
> 
> https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266
> 
> Did you have any errors during that?

No, I'm noob on the subject, so I followed this video:  https://youtu.be/QBj8OLig8Kg

pocketscience, 2020-03-04:
Noobs should follow the documented repository build instructions... :)

renandw, 2020-03-05:
One video that helped me a lot to use Ubuntu on MacBook is this: https://youtu.be/b5VU_UWzeNI

renandw, 2020-03-05:
And also these:
https://youtu.be/kYEzEFH5LiM
https://youtu.be/T-oSjMCmNYk

In some previous issues I found these videos.

MizzyAntunes, 2020-03-05:
> One video that helped me a lot to use Ubuntu on MacBook is this: https://youtu.be/b5VU_UWzeNI

Yes, but i'm a Windows user

renandw, 2020-03-05:
No problem at all

MizzyAntunes, 2020-03-05:
> Noobs should follow the documented repository build instructions... :)

Yes, I know, but I don't know where to run these commands. Like her:

Initialize and sync all submodules (recursively):
git submodule update --init --recursive

Ok, Where i do that?

MizzyAntunes, 2020-03-05:
> No problem at all

wow, is it for Windows too?

closed 2020-03-10 by MizzyAntunes