maximkulkin/esp-homekit-demo #318: How to change de Lock example?

MizzyAntunes, 2020-03-05:
I would like to use the "Lock" model of the examples, but I don't want it to lock itself after a few seconds. How can I change this?

maximkulkin, 2020-03-05:
By modifying the code

pocketscience, 2020-03-05:
That's what I'd do too... maybe starting with disabling the timer that does the auto-locking after unlock_period seconds...

MizzyAntunes, 2020-03-05:
> By modifying the code

I know, but I don't know how to do that, because I'm a noob. I think someone who knows programming would know how to help me

MizzyAntunes, 2020-03-05:
> That's what I'd do too... maybe starting with disabling the timer that does the auto-locking after unlock_period seconds...

Ok, tks

pocketscience, 2020-03-05:
You might want to think about some basic programming courses instead of jumping head-first into embedded development. There's a lot to learn just on the programming side, let alone all the intricacies of embedded systems. If you're after a HomeKit lock in the meantime it might be simpler to buy one... Sorry, not trying to be rude - just people here are going to struggle to help you through every single step - especially when you ask such basic questions as these and others... Help yourself first, then others might be more willing to provide more assistance.

MizzyAntunes, 2020-03-05:
> You might want to think about some basic programming courses instead of jumping head-first into embedded development. There's a lot to learn just on the programming side, let alone all the [...]

ok, I understand, is that my interest was only in the homekit. Excuse me

closed 2020-03-10 by MizzyAntunes