maximkulkin/esp-homekit-demo #319: ESP8266 1MB Menuconfig and Size

mriksman, 2020-03-12:
Hey Community,

To compile, I needed to turn on go to `Component Config | LWIP | Enable IPv6`, otherwise the `mdns.h` would complain that it didn't know `ip6_addr_t`. I also had to enable the `Component Config | mDNS | Enable mDNS`, otherwise the linker couldn't find the `mdns` functions (like `mdns_init()`)

Has anyone else seen this? I haven't seen an example sdkconfig file required to work with esp-homekit.

And how about size? The LED example compiles to 
```
 DRAM .data size:    3128 bytes
 DRAM .bss  size:   11832 bytes
Used static DRAM:   14960 bytes (  83344 available, 15.2% used)
Used static IRAM:   30563 bytes (  18589 available, 62.2% used)
      Flash code:  427796 bytes
    Flash rodata:   77748 bytes
Total image size:~ 539235 bytes (.bin may be padded larger)
```
How could this be shrunk more to enable OTA?

maximkulkin, 2020-03-12:
AFAIK, you need to enable mDNS and IPv6 will be enabled automatically.

As for the size: you can try enabling HOMEKIT_SMALL (Component config -> HomeKit -> Minimize firmware size) at a price of slower pairing. Can help you save ~70Kb last time I checked. But overall, doing OTA on modules with small flash memory (like 1MB) might be tricky.

mriksman, 2020-03-12:
Thanks @maximkulkin. 
I'm surprised that no one has written a nice OTA library that doesn't need to have a 50/50 split (actually, I think HomeACcessoryKid/life-cycle-manager does, but I am not using esp-open-rtos). `ota_0` could be the 'OTA Loader', and `ota_1` could be the main program. In the main program, you could navigate to a page, point to a .bin file on a HTTPS server, and hit GO. `ota_0` would then load and load the .bin straight over the top of `ota_1`.

mriksman, 2020-03-12:
@maximkulkin. You're a smarter guy than I am. Is it possible to load an Arduino based Updater program to `ota_0` and have it upload a .bin file to `ota_1` created with ESP-IDF? I notice Arduino Library uses `esp_ota_set_boot_partition()` so it's feasible I could uses this to switch back and forth.

closed 2020-06-26 by maximkulkin

mriksman, 2020-06-26:
FYI - I managed to do as I said; I have a 320KB OTA boot loader and a 640KB main application partition. To update, I set the file in the web interface, which then boots to the boot loader partition which receives the file. Works well. This also has allowed me to run `esp-homekit` without the need for `HOMEKIT_SMALL` - which greatly improves pairing speed.