maximkulkin/esp-homekit-demo #321: Lock - Siri cannot issue the lock command

wasp100, 2020-03-14:
Hi @maximkulkin 
I'm using the Lock example, and noticed that when I ask Siri to lock the door, the response is
`"Sorry, I wasn't able to lock your Lock. For more details, check the Home app."`

I can however unlock the lock through Siri with no problem!

I just can't seem to get Siri to lock the lock, and I cannot see anything in the code that would fix this.

renandw, 2020-03-15:
How is your code?
In the lock example in the esp-homekit-demo, you unlock and after 5 seconds it locks again.
Does Siri unlock it?

wasp100, 2020-03-15:
I’ve tested this with the original lock example code. When you ask Siri to unlock the lock, the response is “ok, lock is unlocked”. However when you ask Siri to lock the lock, the response is “sorry I’m unable to lock your lock...” 
I’ve modified the code to allow for a separate lock and unlock, but the issue still exists.

wasp100, 2020-04-07:
Hi @maximkulkin 

Can you please assist me in getting this to work? Could the issue lie in the ESPHomeKit code?

My coding seems fine and it works if i toggle the switch in HomeKit. It also works when I ask Siri to unlock. However, if I ask Siri to lock, I get a "sorry I'm unable to lock your lock" response.

Thanks!

maximkulkin, 2020-04-07:
Ok, I have debugged this down to the following problem: when you update LOCK_TARGET_STATE characteristic (of type uint8_t), Home app properly sends numeric value. However, when you ask Siri to lock or unlock a lock, Siri sends a boolean value. While it's kind of nonsense, however I originally coded support for allowing booleans instead of integers, thus the error of invalid value format was not raised, but the value wasn't handled properly and always defaulted to zero.

I have pushed fix to properly convert booleans to integers if needed. In my tests Siri now properly locks and unlocks a lock.

wasp100, 2020-04-07:
Thank you so much @maximkulkin 
Silly question: what’s the easiest way to update esp-HomeKit?

maximkulkin, 2020-04-08:
Just do
```
git pull --rebase origin master
git submodule update --init --recursive
```
since I have updated all versions in esp-homekit-demo repository.

Alternatively, go to components/common/homekit and do `git pull --rebase origin master`

wasp100, 2020-04-08:
@maximkulkin Thank you so much. Just tested it and it works like a charm! Much appreciated.

closed 2020-04-08 by wasp100