maximkulkin/esp-homekit-demo #324: Is homeassistant.io integration possible for these homekit devices

sanilsdevan, 2020-03-20:
I am trying to integrate homekit enabled sonoff_basic to [homeassistant](https://www.home-assistant.io/) . I am using esp8266 and doing proto development. I did an erase_flash & flashed the firmware on my ESP8266. then configured the wifi OTA. Now, my home app in my Ipad is detecting the sonoff_device to be added (i haven't added it because a zero-config is needed for home assistant to integrate device), but when I try to detect it with homeassistant.io, it can't detect any homekit device. I even tried the sample LED example. Still the same result.

The home assistant documentation says it can only detect certain type of devices

[https://www.home-assistant.io/integrations/homekit_controller/](https://www.home-assistant.io/integrations/homekit_controller/)

the device types includes
Alarm Control Panel (HomeKit security system)
Climate (HomeKit thermostats and air conditioners)
Cover (HomeKit garage door openers, windows, or window coverings)
Light (HomeKit lights)
Lock (HomeKit lock)
Switch (HomeKit switches)
Binary Sensor (HomeKit motion, contact, occupancy and smoke sensors)
Sensor (HomeKit humidity, temperature, co2 and light level sensors)
Fan
Air Quality

I believe that the Sonoff_basic is a **Homekit switch**
Also, Sample_LED is **Homekit light.**

please correct me if I am wrong about these categories

maximkulkin, 2020-04-07:
Yes, you're correct: sonoff_basic example exposes a SWITCH accessory and led example is a LIGHTBULB.

Make sure that your accessory is in the same network as you HomeAssistant server.

sanilsdevan, 2020-04-10:
OK, thank you..

May be due to integration problems, the device is not discovered by homeassistant. But no problems when using the home app. I have raised an issue in the home assistant community. no response yet. waiting..

sanilsdevan, 2020-05-02:
I have successfully added sonoff basic to homeassistant & then using the homeassistant bridge to homekit. Now it is interconnected. If I click the switch to ON in homekit, the state of the switch gets updated in the homeassistant. Similarly if i turn off the switch in homeassistant, the same gets updated in homekit too.

But I have an issue with the button example. it gets stuck at mdns announcement & doesn't proceed further. 
`14:48:43.280 -> ESP-Open-SDK ver: 0.0.1 compiled @ Apr 12 2020 20:43:53`
`14:48:43.280 -> phy ver: 273, pp ver: 8.3`
`14:48:43.406 -> `
`14:48:43.406 -> >>> wifi_config: Initializing WiFi config`
`14:48:43.406 -> >>> wifi_config: Connecting to Heartbeatz`
`14:48:43.406 -> Button example`
`14:48:43.406 -> mode : sta(b4:e6:2d:0a:ae:90)`
`14:48:43.406 -> add if0`
`14:48:43.477 -> scandone`
`14:48:44.772 -> add 0`
`14:48:44.772 -> aid 6`
`14:48:44.772 -> cnt `
`14:48:44.912 -> `
`14:48:44.912 -> connected with Heartbeatz, channel 9`
`14:48:44.912 -> dhcp client start...`
`14:48:45.766 -> ip:192.168.0.161,mask:255.255.255.0,gw:192.168.0.1`
`14:48:53.369 -> >>> wifi_config: Connected to WiFi network`
`14:48:53.470 -> >>> HomeKit: Starting server`
`14:48:53.470 -> >>> HomeKit: Using existing accessory ID: 11:87:E7:87:89:B0`
`14:48:53.470 -> >>> HomeKit: Found admin pairing with ca19e569-f4bb-41fd-a970-5e579711cca7, disabling pair setup`
`14:48:53.470 -> >>> HomeKit: Configuring mDNS`
`14:48:53.470 -> mDNS announcement: Name=Button md=MyButtonpv=1.0id=11:87:E7:87:89:B0c#=1s#=1ff=0sf=0ci=15sh=dG3JQQ== Port=5556 TTL=4500`
`
I reflashed the button example & tried adding the button directly to homekit without adding to homeassistant. In this case too it was the same. except that if i go to the settings of the button in homekit, then the mdns announcement proceeds further. Then the button remains connected untill it gets disconnected when i remove power. When i connect it to the power, again it  gets stuck at the mdns announcement and doesnot proceed further untill i open the homekit app button 
settings page
[image]

sanilsdevan, 2020-05-02:
When I go to the button settings and for any action .. for example single press, if I select 'TEST THIS ACTION', then the rest of the pairing continues
[image]

maximkulkin, 2020-05-03:
The problem there is that iOS controllers do not rush to re-establish connection to an accessories after loosing a connection. I also have problems especially with buttons.

sanilsdevan, 2020-05-06:
Oh.... well so there is no choice i guess. 

I was going around finding a solution & i found this guy doing the exactly same thing in **HAP-NodeJS** which helps a button **which uses mqtt connect to homekit**

[https://www.makermoekoe.net/post/a-dashbutton-like-thing-called-octoclick](url)

He has the code in **.ino file which uses mqtt command** written on ESP8266 which is connected to HAP-nodeJS which has mqtt controller installed. The button device turns off instantaneously after it send the mqtt command. But when the button it clicked again, it turn on, sends the mqtt command to HAP-NodeJS & aturns off again. The surprising thing is the button reconnects to the homekit instantaneously without any delays which turn on a light which is also connected to homekit

Is the esp-homekit in any way similar to **HAP-HodeJS?** Just in case if there is a workaround..

maximkulkin, 2020-05-06:
That is not HomeKit.. It's thing called IFTTT which I guess is a MQTT-based solution. That also unlike HomeKit sends your data to third-party cloud services.

closed 2020-06-26 by maximkulkin