maximkulkin/esp-homekit-demo #325: Flash problem (Makefile)

borcares, 2020-03-20:
Hi..
When I try to flash the Esp8266 with the command: _make -C examples/led flash monitor_

I have this error: 
_Makefile:13: /Users/xxx/esp8266/esp-homekit-demo/esp-open-rtos/common.mk: No such file or directory
make: *** No rule to make target `/Users/xxx/esp8266/esp-homekit-demo/esp-open-rtos/common.mk'.  Stop._

hope any1 can help... thank you

maximkulkin, 2020-03-20:
Make sure the file is there: /Users/xxx/esp8266/esp-homekit-demo/esp-open-rtos/common.mk

Otherwise check that SDK_PATH environment variable is properly set.

borcares, 2020-03-20:
Thanks Maxim,

I have been able to flash with the comand:

_esptool.py -p /dev/cu.wchusbserial1420 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x00000 led.bin_

But I think the code doesn't work in my esp... why I don't view the accessory in home app and in serial monitor shows that:

_ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c_

You know what the problem might be?

Thanks in advance.

closed 2020-03-22 by borcares