maximkulkin/esp-homekit-demo #327: Unable to connect accessory

fduman, 2020-03-21:
I try to run examples/led_status but an error occurred:

pp_task_hdl : 3ffeffc0, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3f30, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Mar 21 2020 07:14:05
phy ver: 273, pp ver: 8.3

mode : sta(5c:cf:7f:16:50:80)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x100000
>>> HomeKit: Generated new accessory ID: BD:A0:0C:34:D5:3A
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Sample LEDmd=MyLEDpv=1.0id=BD:A0:0C:34:D5:3Ac#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
scandone
add 0
aid 2
cnt

connected with greenn, channel 6
dhcp client start...
ip:192.168.0.149,mask:255.255.255.0,gw:192.168.0.1
>>> HomeKit: Got new client connection: 4 from 192.168.0.192
>>> HomeKit: [Client 4] Pair Setup Step 1/3
rtos assert /opt/esp-open-rtos/FreeRTOS/Source/timers.c 811
abort() invoked at 0x4020491b.

shaurya0406, 2020-03-22:
I am also unable to connect to esp32 running examples/esp32/led using esp-idf v4.0 latest master branch

this is the error

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7196
load:0x40078000,len:13092
load:0x40080400,len:3936
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.2-dev-792-g6330b3345 2nd stage bootloader
I (29) boot: compile time 20:19:55
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d5c8 (120264) map
I (159) esp_image: segment 1: paddr=0x0002d5f0 vaddr=0x3ffb0000 size=0x02a28 ( 10792) load
I (164) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x918ac (596140) map
0x400d0020: _stext at ??:?

I (392) esp_image: segment 3: paddr=0x000c18d4 vaddr=0x3ffb2a28 size=0x014fc (  5372) load
I (395) esp_image: segment 4: paddr=0x000c2dd8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/shauryachandra/MyProg/ESP32/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (400) esp_image: segment 5: paddr=0x000c31e4 vaddr=0x40080404 size=0x14d44 ( 85316) load
I (457) boot: Loaded app from partition at offset 0x10000
I (457) boot: Disabling RNG early entropy source...
I (457) cpu_start: Pro cpu up.
I (461) cpu_start: Application information:
I (466) cpu_start: Project name:     led
I (470) cpu_start: App version:      9ba5ba5-dirty
I (476) cpu_start: Compile time:     Mar 22 2020 20:19:48
I (482) cpu_start: ELF file SHA256:  7ce307b24ffddcfe...
I (488) cpu_start: ESP-IDF:          v4.2-dev-792-g6330b3345
I (494) cpu_start: Starting app cpu, entry point is 0x40081490
0x40081490: call_start_cpu1 at /Users/shauryachandra/MyProg/ESP32/esp-idf/components/esp32/cpu_start.c:273

I (0) cpu_start: App cpu up.
I (505) heap_init: Initializing. RAM available for dynamic allocation:
I (511) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (517) heap_init: At 3FFB9FC0 len 00026040 (152 KiB): DRAM
I (524) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (530) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (536) heap_init: At 40095148 len 0000AEB8 (43 KiB): IRAM
I (543) cpu_start: Pro cpu start user code
I (561) spi_flash: detected chip: generic
I (562) spi_flash: flash io: dio
W (562) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (572) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (638) wifi:wifi driver task: 3ffc1858, prio:23, stack:3584, core=0
I (638) system_api: Base MAC address is not set
I (638) system_api: read default base MAC address from EFUSE
I (768) wifi:wifi firmware version: b88ceb8
[... 87 lines cut ...]
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 5 value (154 bytes): .WT\xB0,\xE3!0\xED\x94\x1F@\xD4\x98\xA5"\xE0\x94;\x07\xBC\xD5\x11#^\x01\x08L\xD1v\x1C\x9C@t\x00xQ^\xD4\xB1s\x8A\xB2\xE8c\x80?<[\xFE\x07o\x91h\x95M\xBC\x1Cp1%3\x0D\x01\xAE\xA1\xE5\x08\xD4\xAE\xA9\x08n\x92\xB7h\x03z\xDF\xA4\xB7\xD3x$Y\xAEp\xC2\xD5}y}7L\x0D\xD2\xF1\xA2\x8B\xA9\xFA\xD2I\xE7\x8F"\xA0E\x808\xA9\x0CP\xF2G\xA4)\xC3\x13\x8E\x92\x9A'\xAD\x05\x99\xD4\xB0\x93 i94\xB9|\xE6\x11\xF2\x95Q\xFDOp\xA5\xB0\x1Dv\x84\xB5\x04L^\x14
>>> tlv_debug: Type 6 value (1 bytes): \x05
>>> HomeKit: [Client 55] Pair Setup Step 3/3
>>> homekit_server_on_pair_setup: Free heap: 213192
>>> homekit_server_on_pair_setup: [Client 55] Calculating shared secret
>>> homekit_server_on_pair_setup: [Client 55] Decrypting payload
>>> homekit_server_on_pair_setup: [Client 55] Importing device public key
>>> homekit_server_on_pair_setup: [Client 55] Calculating DeviceX
>>> homekit_server_on_pair_setup: [Client 55] Verifying device signature
!!! HomeKit: Failed to compact HomeKit storage: sector data read error
!!! HomeKit: Failed to write pairing info to HomeKit storage: max number of pairings
!!! HomeKit: [Client 55] Failed to store pairing (code -2)
>>> send_tlv_response: [Client 55] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x06
>>> tlv_debug: Type 7 value (1 bytes): \x01
>>> client_send: [Client 55] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 6\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x06\x07\x01\x01
>>> homekit_client_process: [Client 55] Finished processing
>>> HomeKit: [Client 55] Closing client connection

```
Please help
Thank you
Regards

shaurya0406, 2020-03-22:
ok
I just tried running
`make -j4 menuconfig`
 `make -j4 -C led erase_flash flash monitor`

I got it running and I am able to connect my device.

So is there something wrong in executing 
`idf.py menuconfig`
`idf.py erase_flash flash`

maximkulkin, 2020-04-08:
@shaurya0406 I'm using Make to build, so it is guaranteed to work. idf.py is not

shaurya0406, 2020-04-08:
Thank you so much @maximkulkin

closed 2020-06-26 by maximkulkin