maximkulkin/esp-homekit-demo #330: "Unable to join network"; no buf for probe, ie len 0; max connection!

AramVartanyan, 2020-03-29:
I would like to report a strange issue, which I am facing in the last months.

Sometimes I could not connect easily to the AP created by the ESP8266. (It also affects the overall stability of the device after pairing.) 8 out of 10 times I see the iOS message:

Unable to join network "Shelly1-XXXX",

than in the Terminal I got:

no buf for probe, ie len 0

or

max connection!

I have noticed that a re-flashing sometimes works, but not always. And I don't believe that it is a hardware issue, because re-flashing a working device can lead to this issue and is happening on brand new hardware too. (I am using Shelly1 devices and Wemos D1 boards.)

No matter what I have tried, I cannot find the root case of this issue.

I wander if somebody have the same issue or knowing a way to resolve it?!

I am using the latest versions of all components.
This is the main code I am using:
https://github.com/AramVartanyan/esp-homekit-demo/blob/master/examples/shelly1/shelly1.c
but it also appears with unchanged code from the demo:
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/sonoff_basic

So I summarized that the issue is somewhere else.

This is the error log:

l??ll8c??`?'??nl$o???

                     l?ldl?o??o????c
                                    ?
                                     dd$s$?#?og??d?'og?o܄?'??nd?c?'o??d?no??`
                                                                             ????`sd
                                                                                    ?'o??|???n??clpp_task_hdl : 3fff0030, prio:14, stack:512
pm_task_hdl : 3ffef920, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4008, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Mar 21 2020 13:02:02
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
!!! wifi_config: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: Starting DHCP server
mode : sta(e0:98:06:8d:87:c3) + softAP(e2:98:06:8d:87:c3)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
scandone
scandone
no buf for probe, ie len 0
scandone
scandone
add 1
aid 1
station: 84:89:ad:c6:0a:2a join, AID = 1
scandone
scandone
scandone
scandone
scandone
station: 84:89:ad:c6:0a:2a leave, AID = 1
rm match
no buf for probe, ie len 0
no buf for probe, ie len 0
no buf for probe, ie len 0
no buf for probe, ie len 0
scandone
scandone
scandone
scandone
scandone
scandone
max connection!
scandone
add 1
aid 1
station: 84:89:ad:c6:0a:2a join, AID = 1
scandone
no buf for probe, ie len 0
no buf for probe, ie len 0
no buf for probe, ie len 0
scandone
scandone

I also found this:
https://github.com/RavenSystem/esp-homekit-devices/issues/569
and this:
https://github.com/RavenSystem/esp-homekit-devices/issues/63

But they were not so useful.

AramVartanyan, 2020-04-02:
This fix the issue so far.
https://github.com/AramVartanyan/esp-wifi-config/commit/8b8166cd0536b9aa028206e727cd1494e3755ee0

closed 2020-04-02 by AramVartanyan

maximkulkin, 2020-04-06:
Thank you @AramVartanyan for pointers. HTTP server in wifi_config is a bit too simple, being able to handle only a single connection which I guess sometimes becomes an issue. Dropping connection on error certainly makes sense, although you need to be careful to avoid dropping on errors like EAGAIN. I will look into improving overall HTTP server matureness.

AramVartanyan, 2020-04-06:
You are welcome, @maximkulkin. I am glad that my findings are helpful. And I thank you for your commitment in this great project.