maximkulkin/esp-homekit-demo #331: driver for sm2135

Kristian8606, 2020-04-06:
Hello @maximkulkin ! Would it be difficult to i implement this library https://github.com/NURDspace/esphome/tree/sm2135/esphome/components/sm2135 for freeRtos to manage this inexpensive light bulb. https://github.com/RavenSystem/esp-homekit-devices/issues/870

maximkulkin, 2020-04-06:
It's just writing to I2C device registers. All you need to do is to use [two simple commands](https://github.com/NURDspace/esphome/blob/sm2135/esphome/components/sm2135/sm2135.cpp#L60-L77): first one sets white color of given temperature, second one sets arbitrary RGB color, assuming that library actually works. Couldn't find a proper datasheet for that chip, only one in chinese and it didn't seem to have information on registers and commands.

maximkulkin, 2020-04-06:
Ok, found datasheet translated to english: [SM2135E_zh-CN_en-US_translated.pdf](https://github.com/arendst/Sonoff-Tasmota/files/3656603/SM2135E_zh-CN_en-US_translated.pdf). It contains information on format of commands.

Kristian8606, 2020-04-06:
Thank you for your advice.  I will try it and report the result you.

Kristian8606, 2020-04-06:
I looked through the code and realized that I couldn't handle it.

closed 2020-04-06 by Kristian8606