maximkulkin/esp-homekit-demo #332: Homekit storage problem

borcares, 2020-04-07:
Hi..

After flash the led example, in serial monitor shows that:

>>> HomeKit: Starting server
!!! HomeKit: Failed to read HomeKit storage magic
>>> HomeKit: Formatting HomeKit storage at 0x100000
!!! HomeKit: Failed to erase HomeKit storage
>>> HomeKit: Resetting HomeKit storage
!!! HomeKit: Failed to reset HomeKit storage
>>> HomeKit: Generated new accessory ID: 61:CE:30:5E:75:42
!!! HomeKit: Failed to write accessory ID to HomeKit storage
scandone
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to HomeKit storage
>>> HomeKit: Configuring mDNS

...

hope any1 can help... thank you

maximkulkin, 2020-04-07:
Which ESP module (which hardware) are you using?

borcares, 2020-04-07:
nodemcu lolin v3

maximkulkin, 2020-04-07:
Try changing HOMEKIT_SPI_FLASH_BASE_ADDR to a smaller value (e.g. 0x7A000)

borcares, 2020-04-07:
I flash the esp with this command:

esptool.py -p /dev/cu.wchusbserial1420 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin

as far as I understand the HOMEKIT_SPI_FLASH_BASE_ADDR is the -fs 1MB... I've changed it to -fs 8m and I have the same problem.

maximkulkin, 2020-04-08:
No, HOMEKIT_SPI_FLASH_BASE_ADDR has nothing to do with -fs option of esptool.py. You need to recompile firmware with that flag (e.g. `make HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000 all`)

borcares, 2020-04-09:
Thanks a lot!  that was the problem!  congratulations on the code, you are a genius!

closed 2020-04-09 by borcares