maximkulkin/esp-homekit-demo #334: esp32 button example not compiling, issue with wifi init

DBabs1, 2020-04-11:
Hey, I can't get the button example to compile without issue. If I copy the wifi init of the esp32/led example it will compile and load in homekit but the button won't work correctly. the esp32/led compiles fine here. 

here's the 2 errors, any ideas on what i can do to make this work?:

/Users/Derek/esp/esp-homekit-demo/examples/esp32/button/main/main.c:32:21: error: implicit declaration of function 'esp_netif_init'; did you mean 'esp_wifi_init'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_netif_init());
                     ^~~~~~~~~~~~~~
/Users/Derek/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
/Users/Derek/esp/esp-homekit-demo/examples/esp32/button/main/main.c:35:5: error: implicit declaration of function 'esp_netif_create_default_wifi_sta' [-Werror=implicit-function-declaration]
     esp_netif_create_default_wifi_sta();

yz88, 2020-04-13:
Hi, first of all thanks for providing a button example for esp32. I'm facing the same issue with a fresh esp-idf toolchain v3.3.1 on Windows and unchanged code from the button example.

```bash
CC examples/esp32/button/build/lwip/lwip/src/netif/lowpan6.o
CC examples/esp32/button/build/lwip/lwip/src/netif/ethernet.o
CC examples/esp32/button/build/lwip/lwip/src/netif/slipif.o
CC examples/esp32/button/build/lwip/port/esp32/vfs_lwip.o
CC examples/esp32/button/build/lwip/port/esp32/freertos/sys_arch.o
CC examples/esp32/button/build/lwip/port/esp32/netif/ethernetif.o
CC examples/esp32/button/build/lwip/port/esp32/netif/dhcp_state.o
CC examples/esp32/button/build/lwip/port/esp32/netif/wlanif.o
CC examples/esp32/button/build/lwip/port/esp32/debug/lwip_debug.o
AR examples/esp32/button/build/lwip/liblwip.a
make[1]: Verzeichnis „/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/build/lwip“ wird verlassen
make[1]: Verzeichnis „/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/build/main“ wird betreten
CC examples/esp32/button/build/main/main.o
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c: In function 'event_handler':
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:22:23: error: 'WIFI_EVENT' undeclared (first use in this function)
     if (event_base == WIFI_EVENT && (event_id == WIFI_EVENT_STA_START || event_id == WIFI_EVENT_STA_DISCONNECTED)) {
                       ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:22:23: note: each undeclared identifier is reported only once for each function it appears in
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:22:50: error: 'WIFI_EVENT_STA_START' undeclared (first use in this function)
     if (event_base == WIFI_EVENT && (event_id == WIFI_EVENT_STA_START || event_id == WIFI_EVENT_STA_DISCONNECTED)) {
                                                  ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:22:86: error: 'WIFI_EVENT_STA_DISCONNECTED' undeclared (first use in this function)
     if (event_base == WIFI_EVENT && (event_id == WIFI_EVENT_STA_START || event_id == WIFI_EVENT_STA_DISCONNECTED)) {
                                                                                      ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:25:30: error: 'IP_EVENT' undeclared (first use in this function)
     } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
                              ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:25:54: error: 'IP_EVENT_STA_GOT_IP' undeclared (first use in this function)
     } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
                                                      ^
In file included from C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-idf/components/esp32/include/esp_wifi.h:62:0,
                 from C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:2:
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c: In function 'wifi_init':
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:32:21: error: implicit declaration of function 'esp_netif_init' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_netif_init());
                     ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:35:5: error: implicit declaration of function 'esp_netif_create_default_wifi_sta' [-Werror=implicit-function-declaration]
     esp_netif_create_default_wifi_sta();
     ^
In file included from C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-idf/components/esp32/include/esp_wifi.h:62:0,
                 from C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:2:
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/main/main.c:37:48: error: 'WIFI_EVENT' undeclared (first use in this function)
     ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
                                                ^
C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
[... cut ...]
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/Holger/esp-idf-v3.3.1/esp-idf/make/component_wrapper.mk:290: main.o] Fehler 1
make[1]: Verzeichnis „/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button/build/main“ wird verlassen
make: *** [C:/Users/wiege/Documents/esp32-dev-env/msys32/home/Holger/esp-idf-v3.3.1/esp-idf/make/project.mk:552: component-main-build] Fehler 2
make: Verzeichnis „/home/Holger/esp-idf-v3.3.1/esp-homekit-demo/examples/esp32/button“ wird verlassen

```

rillhu, 2020-04-13:
> Hey, I can't get the button example to compile without issue. If I copy the wifi init of the esp32/led example it will compile and load in homekit but the button won't work correctly. the esp32/led [...]

Hi, This esp32 example is a little old-fashioned, you could refer to my repository to rewrite the mian.c.(although I used another homekit library from younghyunjo, you could still find useful style of main.c from my repo). 
You can find my [main.c](https://github.com/rillhu/esp32-homekit/blob/master/examples/switch/main/hap_main.c) here.

kunjal83, 2020-05-31:
Use the current example and add the below as marked to compile successfully. 

#include <freertos/FreeRTOS.h>
_++<Add New Line>_
_#include <freertos/event_groups.h>_
_++<Add New Line>_
#include <freertos/task.h>

#define EXAMPLE_ESP_WIFI_SSID      "xxxxx"
#define EXAMPLE_ESP_WIFI_PASS      "xxxxx"
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

Replace exiting method for wifi_init with below: 

_static void wifi_init() {
    s_wifi_event_group = xEventGroupCreate();
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI("Tag", "wifi_init_sta finished.");
    ESP_LOGI("Tag", "connect to ap SSID:%s password:%s",
             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);

}_

maximkulkin, 2020-06-04:
@DBabs1 I've re-checked button example and it compiles OK with latest ESP-IDF with both `make` and `idf.py build`

@kunjal83 Not sure where you get this best practices from, but your code is a bit outdated.

kunjal83, 2020-06-05:
Yes, i know i am using the older piece of code. 
I just checked again with the ESP-IDF v4.0.1. i still get error. Could be due to version: 
```
error: implicit declaration of function 'esp_netif_init'; did you mean 'esp_wifi_init'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_netif_init());

```

maximkulkin, 2020-06-09:
Try using latest master

kunjal83, 2020-06-27:
@maximkulkin , yes works with latest master.

closed 2020-11-07 by maximkulkin