maximkulkin/esp-homekit-demo #336: Can you provide more demos

CHNleing, 2020-04-14:
I want to learn more examples

rillhu, 2020-04-15:
> I want to learn more examples
I think current repo provides sufficient ESP8266 examples.  Do you mean ESP32 examples? if you want to see maore ESP32 examples, you could refer to my repository (although I used another homekit library from younghyunjo, you could still find useful examples from my repo).
You can find my [esp32-homekit](https://github.com/rillhu/esp32-homekit) here.

CHNleing, 2020-04-15:
I want to know more types of cases, such as air conditioners, windows, thermostats

maximkulkin, 2020-04-15:
@CHNleing download the HomeKit Specification and indulge yourself.

CHNleing, 2020-04-15:
@maximkulkin Thank you very much, I found Homekit Accessory Protocol Specification from Apple official website.
I am trying to read it.

closed 2020-04-15 by CHNleing