maximkulkin/esp-homekit-demo #337: SonOff switch button with timer

ricou27, 2020-04-17:
Hello,

I would like when I push the button to activate the SonOff contact, de-activate it after 1 sec. With the Sonoff_basic example, I tried to add some code to the "button_event_single_press" switch case like :
            switch_on.value.bool_value = !switch_on.value.bool_value;
            relay_write(switch_on.value.bool_value);
            homekit_characteristic_notify(&switch_on, switch_on.value);
            vTaskDelay(1000 / portTICK_PERIOD_MS);
            switch_on.value.bool_value = !switch_on.value.bool_value;
            relay_write(switch_on.value.bool_value);
            homekit_characteristic_notify(&switch_on, switch_on.value);

but impossible to do the job.

Have you an idea on how to do that

Thanks

maximkulkin, 2020-04-19:
You can not just do vTaskDelay inside button callback, you're blocking from inside interrupt which might mess up the whole OS.

Instead you need to set up a timer to switch your thing back. Check out [lock example](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/lock/main.c#L199-L208), where (if you set unlock_period setting) lock is opened only for small duration and then locked back.

ricou27, 2020-04-20:
Many thanks for your answer.  That’s Work by using a timer. Regards

closed 2020-04-20 by ricou27