maximkulkin/esp-homekit-demo #339: Failed to compact HomeKit storage: sector data read error

pocketscience, 2020-04-22:
I've just started seeing the following when pairing after building the new esp-button based button example. Does this indicate a hardware error? The device (Sonoff Basic R2) was previously working fine...

18:12:22.832 -> >>> HomeKit: Got new client connection: 4 from 192.168.1.60
18:12:22.832 -> >>> HomeKit: [Client 4] Pair Setup Step 1/3
18:12:36.820 -> >>> HomeKit: [Client 4] Pair Setup Step 2/3
18:12:50.888 -> >>> HomeKit: [Client 4] Pair Setup Step 3/3
18:12:51.063 -> !!! HomeKit: Failed to compact HomeKit storage: sector data read error
18:12:51.063 -> !!! HomeKit: Failed to write pairing info to HomeKit storage: max number of pairings
18:12:51.063 -> !!! HomeKit: [Client 4] Failed to store pairing (code -2)
18:12:51.284 -> >>> HomeKit: [Client 4] Closing client connection

 Using the following to flash:

#!/bin/bash

SONOFF_PORT="/dev/cu.usbserial-A50285BI"

esptool.py \
            -p $SONOFF_PORT \
            --baud 115200 \
            write_flash \
            -fs 1MB \
            -fm dout \
            -ff 40m \
            0x0 ./firmware/rboot.bin \
            0x1000 ./firmware/blank_config.bin \
            0x2000 ./firmware/new_button.bin

maximkulkin, 2020-04-22:
What flash address you have configured HomeKit settings to be at? Try changing it to a different address.

pocketscience, 2020-04-22:
Good question - where is that configured? I can't recall..

pocketscience, 2020-04-22:
HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000

pocketscience, 2020-04-22:
Hmm, changed to 0x70000 and still get the same error..

maximkulkin, 2020-04-22:
Logs please

pocketscience, 2020-04-22:
Weird. Rebuilt everything with logging enabled and the accessory added just fine to Home... Will do some playing and close this if everything seems ok...

pocketscience, 2020-04-22:
OK closing. Everything seems fine now. Even changed the base address back to 0x7A000 and it's all good.

closed 2020-04-22 by pocketscience

maximkulkin, 2020-04-22:
Keep in mind that modifying Makefile does not automatically rebuild all components (e.g. esp-homekit component), so you need to do clean build (`make clean && make all`)

pocketscience, 2020-04-22:
Yes, I've started using Docker for a change and made some poor assumptions! :)