maximkulkin/esp-homekit-demo #341: Changing GPIO ports/adding several devices

joules22, 2020-04-29:
How can i change the gpio ports so how that i want it? Just change gpio in main.c?

Can i add more devices controlled by one esp?

maximkulkin, 2020-04-29:
So, the general idea is: esp-homekit is a framework for creating firmwares for various devices. So, you take it and develop your own firmware and customize it to whatever your heart desire. Although, this might be a complicated software for beginners as it requires quite a few things to set up. So, this repo holds examples of typical firmwares so you can use them as a starting point. Some people find it sufficient to use off-the-shelf (off-the-repo) examples as is on their devices though.

If you're not a programmer (or do not want to mess with all that software stuff), I would recommend looking at [HAA](https://github.com/RavenSystem/haa) - a third-party universal firmware based on esp-homekit that you can customize to do particular stuff.

joules22, 2020-04-30:
Thank you for your answer! I read about haa before but i dont like the code with only letters. Well i am used to c/c++ but i dont find a good code with documentation. I struggle with the library functions. I can read the code good but i dont know how to write it with the functions, i Is there a well documented code or a good manual? I would like to configure the esp ports to switch the door lock or a light bulb or read about temperature data :)

maximkulkin, 2020-05-09:
Dealing with GPIOs is outside of scope of this framework. Handling GPIOs and interfacing with external hardware is a generic topic and you should look e.g. materials on Arduino (but keep in mind that APIs in esp-open-rtos will be a bit different.

closed 2020-05-09 by joules22