maximkulkin/esp-homekit-demo #343: wifi_config: No configuration found

philhar88, 2020-05-07:
I am not able to get my Sonoff SV (ESP8266) to connect to my WiFi network... It worked once when I first tried, but then every subsequent flash does not. 

my wifi.h
```
#define WIFI_SSID "xxx"
#define WIFI_PASSWORD "xxx"
```

Logs
```
make -C examples/lock test
make: Entering directory '/home/xxx/esp-homekit-demo/examples/lock'
esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
	0x0 /home/xxx/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /home/xxx/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/lock.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0320
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (78.0 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (78.0 kbit/s)...
Writing 454656 @ 0x2000... 454656 (100 %)
Wrote 454656 bytes at 0x2000 in 39.6 seconds (91.8 kbit/s)...
Leaving...
/home/xxx/esp-open-rtos//utils/filteroutput.py --port /dev/ttyUSB0 --baud 115200 --elf ./build/lock.out
Opening /dev/ttyUSB0 at 115200bps...
Booting rom 0.
,�+Zꨥ��UU������+�B-+R�UTE�Ҩ��*�-UU��QB��Ք�K��ZjUQT��ҪUZҰ������0x3fff4180, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May  7 2020 11:56:48
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
!!! wifi_config: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: Starting DHCP server
mode : sta(cc:50:e3:02:9c:f9) + softAP(ce:50:e3:02:9c:f9)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
scandone
scandone
scandone
scandone
scandone
scandone
```

Things I have tried:
- erase_flash
- writing all 0s to the flash:
  `esptool.py -p /dev/ttyUSB0 write_flash -fs 8m 0x00000 blank_1MB.bin`
- writing all 0s to the flash:
- flashing led example (worked once briefly)

Any ideas? It appears the wifi configuration is being loaded but not able to be read?

vsaintloup, 2020-05-07:
Hello @philhar88 ,

I had the same problem and I fixed it by playing with the `#include` tags as well as changing the `wifi_config_init` for `wifi_config_init2` at the bottom of the document (I'm quite new at this, but, I think that not all the demo files were updated with the correct "includes" and "wifi_config" settings, so you sometimes gotta play a little bit with the code and compare it with other demo files templates to make it work).

If you want to do some copy-paste; here is my working example of the code that I tweaked for a Temperature & Motion multi-sensor, take a closer look at `#include <wifi_config.h>` at the beginning and at where you'll find the `wifi_config_init2` at the end. 

```
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>

#include <dht/dht.h>

#ifndef SENSOR_PIN
#error SENSOR_PIN is not specified
#endif

void temperature_sensor_identify(homekit_value_t _value) {
    printf("Temperature sensor identify\n");
}

homekit_characteristic_t temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
homekit_characteristic_t humidity    = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);

void occupancy_identify(homekit_value_t _value) {
    printf("Occupancy identify\n");
}

homekit_characteristic_t occupancy_detected = HOMEKIT_CHARACTERISTIC_(OCCUPANCY_DETECTED, 0);

void temperature_sensor_task(void *_args) {
    gpio_set_pullup(SENSOR_PIN, false, false);

    float humidity_value, temperature_value;
    while (1) {
        bool success = dht_read_float_data(
            DHT_TYPE_DHT11, SENSOR_PIN,
            &humidity_value, &temperature_value
        );
        if (success) {
            temperature.value.float_value = temperature_value;
            humidity.value.float_value = humidity_value;

            homekit_characteristic_notify(&temperature, HOMEKIT_FLOAT(temperature_value));
            homekit_characteristic_notify(&humidity, HOMEKIT_FLOAT(humidity_value));
        } else {
            printf("Couldnt read data from sensor\n");
        }

        vTaskDelay(3000 / portTICK_PERIOD_MS);
    }

void temperature_sensor_init() {
    xTaskCreate(temperature_sensor_task, "Temperatore Sensor", 256, NULL, 2, NULL);
}

void sensor_callback(bool high, void *context) {
    occupancy_detected.value = HOMEKIT_UINT8(high ? 1 : 0);
    homekit_characteristic_notify(&occupancy_detected, occupancy_detected.value);
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_thermostat, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Temperature Sensor"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Vinou inc."),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "12345"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyTemperatureSensor"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, temperature_sensor_identify),
            NULL
        }),
        HOMEKIT_SERVICE(TEMPERATURE_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Temperature Sensor"),
            &temperature,
            NULL
        }),
        HOMEKIT_SERVICE(HUMIDITY_SENSOR, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Humidity Sensor"),
            &humidity,
            NULL
        }),
        NULL
    }),
    HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_sensor, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Room Sensor"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "1"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Room Sensor"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, occupancy_identify),
            NULL
        }),
        HOMEKIT_SERVICE(OCCUPANCY_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Room Sensor"),
            &occupancy_detected,
            NULL
        }),
        NULL
    }),
    NULL
};

static bool homekit_initialized = false;
static homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void on_wifi_config_event(wifi_config_event_t event) {
    if (event == WIFI_CONFIG_CONNECTED) {
        if (!homekit_initialized) {
            homekit_server_init(&config);
            homekit_initialized = true;
        }
    }
}

void user_init(void) {
    uart_set_baud(0, 115200);
    wifi_config_init2("temperature-sensor", NULL, on_wifi_config_event);
    temperature_sensor_init();
    homekit_server_init(&config);
    if (toggle_create(SENSOR_PIN, sensor_callback, NULL)) {
        printf("Failed to initialize sensor\n");
    }
}
```

I may not have made perfect patches (someone with more programming skills might find some inconsistencies) but now the Wifi on my multi-sensor works perfectly with no errors haha

I believe you'll need to do the erase_flash thing again before re-flashing your unit with a new code since it has to do with wifi configuration (again, not a pro here!)

Hope it helps!

maximkulkin, 2020-05-07:
@philhar88 different examples have different way to configure WiFi. When I started creating examples, a simple hardcoded approach was appropriate. Later I build wifi_config library which allows configuring WiFi network on the device itself, no hardcoding. So, led example uses hardcoded WiFi info from wifi.h, but lock example first creates it’s own WiFi network which you can connect to and you’ll be presented with a page to configure WiFi network to connect accessory to.

@vsaintloup The difference between wifi_config_init and wifi_config_init2 is the type of callback function you can specify: with former it’s just no argument function that is called when connected to a WiFi network, while later accepts an event argument and allows to react on both connected and disconnected events.

philhar88, 2020-05-07:
Thanks @maximkulkin I guess I should have read the code :)

closed 2020-05-07 by maximkulkin