maximkulkin/esp-homekit-demo #345: lightbulb relay

QDUNI, 2020-05-12:
Hi, 

Im trying to create a lightbulb relay in Homekit. I want the relay to turn on for 12 sec if I set the lightbulb on and off 12 sec on an other relay. For brightness I want the relay to turn on for a part of these 12 sec. 
12/100 * brightness. I already wrote the software. but now im thinking is this the best way ? and is it safe to toggle the relay with brightness...? 

thnx!

maximkulkin, 2020-05-12:
What is the use case for such lightbulb? Is light bulb an AC bulb? By saying brightness do you mean implementing dimming? If so, relays are not the right way to implement dimming for an AC lightbulb. You will need a zero crossing detector + triac circuit (google for AC dimmer e.g. [this](https://www.googleadservices.com/pagead/aclk?sa=L&ai=DChcSEwjQjLjCwa7pAhXWIK0GHYynANEYABAHGgJwdg&ohost=www.google.com&cid=CAESP-D2g4pr_Op-CkNq-CFWcSjwUpLS8-7ZTXwjiGXYF-fQGNZDkCYrUEzT3sN4fOW1MqSK8hS1EdrwevIkrktorw&sig=AOD64_05Kgwi4cc7LkQUaeLLjaQQUBHwAA&ctype=5&q=&ved=2ahUKEwi-6azCwa7pAhV2IjQIHYIKCS8Qwg96BAgOEH4&adurl=)

QDUNI, 2020-05-12:
its my fireplace. I can switch it on and of by triggering a relay on for a few seconds. why is a relay no good for this?

maximkulkin, 2020-05-12:
Is it a gas fireplace with electric valve and igniter? Is it designed to be turned on and off that frequently? How does brightness control fits into this?

QDUNI, 2020-05-12:
Its designed with a 433 remote. you can increase the flame by holding the flame button. and decrease by holding small flame button. On the controller unit is a panel connector for home automation. If you wire pin 1 with 3. the flame will go up. if you wire 1 with 2 the flame will go down. Now I want the brightness to set the fireplace flame on a certain height.  By wiring pin 1 and 3 for a amount you can determine the height of the flame

QDUNI, 2020-05-13:
I Got some problems in the code.... if I turn on the device the device will switch call this function

`void relay_write(int relay, int msec)
{
    gpio_write(relay, 1);
    vTaskDelay(msec / portTICK_PERIOD_MS);
    gpio_write(relay, 0);
}`

turning on takes 12 sec this is to much for Homekit to handle? the relay will turn on for 12 sec and then turn of. Great! but Homekit shows a spinning wheel and say not responding.... if I refresh Homekit by switching to another room and go back to the room with the device it will show the correct state. How can I Fix this issue? is the problem due to vTaskDelay(msec / portTICK_PERIOD_MS); ?

maximkulkin, 2020-05-14:
This is an incorrect way to do that, because this code is most likely called directly from characteristic callback which blocks the entire HomeKit server thread for 12 seconds and thus HomeKit server can not respond to iOS controller (e.g. iPhone) that characteristic update was successful (or perform any other processing).

The correct way to do that is to update state and then arm a timer with timeout callback that will deactivate whatever needs to be deactivated (in this case, set GPIO to 0). You can take a look at [lock example](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/lock/main.c#L206-L207) where lock is unlocked for certain period (e.g. several seconds) and then locked back again.

QDUNI, 2020-05-14:
#include <etstimer.h>
#include <esplibs/libmain.h>
ETSTimer Relay_timer;
`void relay_write(int relay, int msec)
{
    gpio_write(relay, 1);
    sdk_os_timer_disarm(&Relay_timer); // This makes sure the the timer is stopped? 
    sdk_os_timer_arm(&Relay_timer, msec, 0); //This starts the timer?
    gpio_write(relay, 0);
}`

Is there anything else I need?

maximkulkin, 2020-05-14:
You need to read that example more

QDUNI, 2020-05-14:
hahaha oke... you mean I have to notify Homekit to be set to on or off before using the timer?

renandw, 2020-05-14:
> You will need a zero crossing detector + triac circuit (google for AC dimmer e.g. [...]

Hi, does de sonnoff_basic_pwm example works with the zero-crossing detector?
thanks

maximkulkin, 2020-05-16:
@QDUNI you need to look for timer callback and how to setup timer callback. What you're missing is that those disarm+arm lines do not magically fix everything - pause execution, return control to server and then unpause when timer triggers and continue from that point. You need to put your `gpio_write(relay, 0)` into a separate function which will be called when timer expires.

@renandw no, i do not think so. For my personal needs I've built a library for that - [esp-ac-dimmer](https://github.com/maximkulkin/esp-ac-dimmer) (keep in mind that if you plan to use it with Triac-based dimmer, the only dimmer type that will work is forward phase dimmer because of the way Triac works).

QDUNI, 2020-05-16:
I already fixed it ! It works amazing thnx!

closed 2020-05-16 by QDUNI