maximkulkin/esp-homekit-demo #346: Flash size arguments in megabits like '8m' are deprecated.

Romadhonfajar, 2020-05-15:
Hello guys, I am a newbie. I'm use mac to flash the sonoff basic. And this is the error that I get when I run flash with this code : make -C examples/sonoff_basic flash monitor

Can you all help Me? 🙏🏼

esptool.py -p /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
		0x0 /Users/romadhonfajar/homekittest/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/romadhonfajar/homekittest/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/sonoff_basic.bin 
WARNING: Flash size arguments in megabits like '8m' are deprecated.
Please use the equivalent size '1MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.8
Serial port /dev/tty.SLAB_USBtoUART
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/tty.SLAB_USBtoUART'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 3201, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 3194, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 273, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/usr/local/bin/esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/local/lib/python3.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.SLAB_USBtoUART: [Errno 2] No such file or directory: '/dev/tty.SLAB_USBtoUART'
make: *** [flash] Error 1

maximkulkin, 2020-05-15:
You need to set environment variable ESPPORT to point to your ESP8266 tty device. Right now it is set or defaults to /dev/tty.SLAB_USBtoUART which seem not to be correct in your case.

The easiest way to find out the correct port is to disconnect your device from USB, do `ls /dev/tty.*`, then reconnect device, do ls again and find out which new device has appeared.

Romadhonfajar, 2020-05-15:
Oh, I see. Now I can figure it out Sir 🙏🏼 My port appears now and not face the error like above.

But on step 12 with the function `make -C examples/sonoff_basic flash monitor`

I get this message Sir 

> esptool.py -p /dev/tty.usbserial --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
> 		0x0 /Users/romadhonfajar/homekittest/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/romadhonfajar/homekittest/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/sonoff_basic.bin 
> 

> WARNING: Flash size arguments in megabits like '8m' are deprecated.
> Please use the equivalent size '1MB'.
> Megabit arguments may be removed in a future release.
> esptool.py v2.8
> Serial port /dev/tty.usbserial
> Connecting........_____....._____....._____....._____....._____....._____....._____
> 
> A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
> make: *** [flash] Error 2

And stuck cannot flash.

Can you help me, Sir? Thanks for your help.

peros550, 2020-05-15:
You are making progress :) 

Now you need to put soNoff in flashing mode. 

The moment you connect your Sonoff to your computer, you need to push its push button . Keep it pushed for 1-2 second and then leave it. 

Then try again with flashing command

Romadhonfajar, 2020-05-15:
> You are making progress :)
> 
> Now you need to put soNoff in flashing mode.
> 
> The moment you connect your Sonoff to your computer, you need to push its push button . Keep it pushed for 1-2 second and then leave it.
> 
> Then try again with flashing command

Thanks you Sir 🙏🏽

Actually I have done it Sir, press the button for 3 seconds on sonoff while connect to my laptop.

maximkulkin, 2020-05-16:
It’s not about how long you press it. When you power it up, bootloader (startup code) checks if button is pressed (or corresponding pin is connected to ground), then it loads special “flashing” code, otherwise it loads normal firmware.

Romadhonfajar, 2020-05-16:
> It’s not about how long you press it. When you power it up, bootloader (startup code) checks if button is pressed (or corresponding pin is connected to ground), then it loads special “flashing” code, otherwise it loads normal firmware.

The button is pressed before I attach the usb serial to my laptop Sir. 
I’m refer to this step on internet : 

> Do not plug into the USB on your computer.
> Do not connect the Sonoff to mains power.
> Press and hold the push button on the Sonoff board.
> Insert the FTDI converter USB in your computer (while holding the push button).
> After 2–3 seconds, release the push button.
> Now it's in flash mode.

But my sonoff cannot flash. Or I miss another step?

Romadhonfajar, 2020-05-16:
> The button is pressed before I attach the usb serial to my laptop Sir.
> I’m refer to this step on internet :
>
>
> But my sonoff cannot flash. Or I miss another step?

Sir @maximkulkin I’ve done with the load for flashing firmware. But I’ve face another problem. 
When I run the program `make -C examples/sonoff_basic flash monitor`

This error came out :

`esptool.py v2.8
Serial port /dev/tty.usbserial
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 3201, in <module>
    _main()
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 3194, in _main
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 273, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/serialposix.py", line 435, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument') `

Now whats my problem Sir? Need your help. And I’m so sorry for my problem with flashing 🙏🏽

kenthinson, 2020-05-26:
@Romadhonfajar looks like your esp tool is not installed correctly.

maximkulkin, 2020-06-04:
That looks like something with FTDI driver. Can you try using corresponding /dev/cu.* interface instead (/dev/cu.usbserial?)?

kenthinson, 2020-06-05:
I had a similar error message as him. I had to reinstall esptool and python pip on my mac. After that it would work. @Romadhonfajar if you run esptool without any arguments does it run without error or does it crash with similar output?

closed 2020-11-07 by maximkulkin