maximkulkin/esp-homekit-demo #348: Question: How to configure a string variable at runtime

peros550, 2020-05-19:
This is not an issue with esp-homekit-demo, but a question for a coding advice/direction. 

I'm building a [multi sensor & AC accessory](https://github.com/peros550/esp-homekit-multiple-sensors) based on your libraries. Except the homekit functionality, one additional function is the ability to send temperature/humidity values at thingspeak.com online IoT database service.  

The accessory sends the data by doing http GET calls at specific intervals. Before I flash the accessory, I have to put in the code the correct API key (string)  which is being transmitted along with the http call. The API string is personal and is related to my account at thingsspeak.com. 

What would be the best way to set a string variable at some point during runtime (once) and use it through out the program's life?

maximkulkin, 2020-05-20:
For larger configuration I would definitely opt for an HTTP server and configuration API. For just one or two values this seems like too much. I would recommend just hardcoding it.

One other approach I was working on is to be able to download data from sysparams storage, parse it, modify it, pack and flash it back. This is totally doable and not hard at all, but requires digging into sysparams source code, and putting together some tool (e.g. based on esptool) which requires some skills. I might write such tool at some point, but no estimates yet.

peros550, 2020-05-20:
thank you for your advices Maxim!

peros550, 2020-06-01:
@maximkulkin Instead of creating an http server, do you think your library esp-wifi-config could be adapted to provide this functionality? A similar approach is used in HAA If I'm not mistaken.

maximkulkin, 2020-06-01:
@peros550 Up to you, but I would never do it that way for my projects. esp-wifi-config, as name suggests, is for configuring WiFi. If you want to do something other than that, it would be some kind of generic configuration UI, which poses question "how to do it in an extensible and reusable way, not too bloated for embedded?". You will probably end up with something not too reusable, but good enough for particular project - thus, a custom HTTP server as I suggested previously.

closed 2020-06-26 by maximkulkin