maximkulkin/esp-homekit-demo #349: Esp01 doesn't boot

brisbaneeastcoast, 2020-05-20:
Hey, 

I'm trying to run the led-sketch on my esp 01S.
After flashing the program everthing works fine. 
When I supply the esp with power after removing it before, 
the esp does not come up. 

When I run the same sketch on my Esp8266 (NodeMCU Amica) it works fine even after reboot.

Then I just commented out the line: "homekit_server_init(&config);"
in the user_init function. 
Afterwards the esp 01S booted normally. 

Can anyone imagine the purpose?

Thank you for helping me!

maximkulkin, 2020-05-20:
Did you pull chip enable and reset pins high?

brisbaneeastcoast, 2020-05-23:
Yes I pulled RST, EN and GPIO0 on high for start. 
But it didn’t work either. 
I think it‘s a software problem because with the same hardware set-up it‘s working when I just leave the "homekit_server_init(&config);" 
commented.

Has anyone ever tried to set up a HomeKit-server on a esp 01?

maximkulkin, 2020-05-23:
I have a HomeKit accessory running on ESP-01 just fine

brisbaneeastcoast, 2020-05-25:
I tried nearly everything but it didn’t work for me. 
Could you tell me which pins you‘ve pulled high while starting the program after flashing ? 
I think that would help. 

Another solution would be to remove and add the accessory with every start of the esp new. 
For that I would like to know when the user_init - function gets called. 
With every startup of the esp ? 

Thank you !

maximkulkin, 2020-06-04:
[image]

@brisbaneeastcoast Just connect 3.3v, RST, CH_EN and GPIO0 to 3.3v rail and GND to GND rail.

> Another solution would be to remove and add the accessory with every start of the esp new.

Mmm, yeah, a "solution". Good luck with that.

closed 2020-06-26 by maximkulkin