maximkulkin/esp-homekit-demo #350: Sonoff Sensor AM2301 Question

kiralikbeyin, 2020-05-22:
@maximkulkin 
I can not read data from original sonoff sensor am2301.
````
//tried
gpio_set_pullup(SENSOR_GPIO, false, false);
gpio_set_pullup(SENSOR_GPIO, true, true);
````
Sensor is working with ewelink (sonoff app)

Any idea?

maximkulkin, 2020-05-22:
gpio_set_pullup() is not how you read data. Can you provide full code?

AramVartanyan, 2020-05-22:
This is what I am using for Sonoff TH16 and it works perfect over a year:
[https://github.com/AramVartanyan/esp-homekit-demo/tree/master/examples/switch_th16](url)

kiralikbeyin, 2020-05-22:
> gpio_set_pullup() is not how you read data. Can you provide full code?

````
void temperature_sensor_task(void *pvParameters)
{
  gpio_set_pullup(SENSOR_GPIO, false, false);

  while (1) {

    if (dht_read_float_data(DHT_TYPE_DHT22, SENSOR_GPIO, &humidity_value, &temperature_value)) {
      printf(">>> Sensor: temperature %g, humidity %g\n", temperature_value, humidity_value);

      if (temperature_value != old_temperature_value)
      {

        if ( ((temperature_value * 1.018) > old_temperature_value) || ((temperature_value * 0.982) < old_temperature_value) )
        {

          old_temperature_value = temperature_value;
          current_temperature.value = HOMEKIT_FLOAT(temperature_value);
          homekit_characteristic_notify(&current_temperature, current_temperature.value);

          old_humidity_value = humidity_value;
          current_humidity.value = HOMEKIT_FLOAT(humidity_value);
          homekit_characteristic_notify(&current_humidity, current_humidity.value);
          update_state();
        }
      }
    }
 
    vTaskDelay(POLL_PERIOD / portTICK_PERIOD_MS);
    taskYIELD();
  }
}
````

DS18B20 works fine.
AM2301 (Alsair brand) works fine.
Original Sonoff AM2301 sensor makes trouble.

I think it is related with dht lib because DHT21 & DHT22 are same in esp-open-rtos but AM2301 is DHT21 (see adafruit dht lib )

kiralikbeyin, 2020-05-23:
@maximkulkin 
New original sonoff sensor (v1.0) is different. (not am2301)
Chip is BB1 0 F 2 G A

Found this:
https://electronics.stackexchange.com/questions/443364/itead-sonoff-rf-wifi-433mhz-switch-identifying-an-unknown-chip-on-the-433mhz

https://au.mouser.com/datasheet/2/368/EFM8BB1-DataSheet-958779.pdf

Is there any way to get temp-hum from new sensor?

[image]

Old one is blue v0.03 which works - it is really am2301

[image]
[image]

maximkulkin, 2020-06-04:
i think this is a place where you pull out your logic analyzer device and start recording what's happening on a wire.

kiralikbeyin, 2020-06-04:
> i think this is a place where you pull out your logic analyzer device and start recording what's happening on a wire.

I bought normal ds18b20 sensors and give up. I will try with logic analyzer later. 

Thank you for the idea.

closed 2020-06-04 by kiralikbeyin