maximkulkin/esp-homekit-demo #352: Adjustment in program required for Sonoff Basic R3

sanilsdevan, 2020-05-22:
Sonoff basic R3 (the latest sonoff basic) has wifi indicator led on GPIO13 and also another indicator led connected to relay on GPIO12. Is there any chance that the program can be modified so that the wifi indicator led turns off (or keeps blinking) if the device gets disconnected from wifi network? Also, the wifi indicator keeps on blinking at the initial stage until the device is configured & gets connected to the wifi network for the first time..thanks

maximkulkin, 2020-05-23:
I have published [my internal version](https://github.com/maximkulkin/esp-sonoff-outlet) of Sonoff S20/S26 firmware, check it out.

sanilsdevan, 2020-05-23:
Thank You so much

closed 2020-05-23 by sanilsdevan