maximkulkin/esp-homekit-demo #353: Single example for BME280 sensor for temp, pressure and humidity

CodezInfiniti, 2020-05-24:
Can you please create an example for BME280 Sensor for temp, pressure and humidity values

renandw, 2020-05-24:
read this: https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve

but you can only get the bin file.

CodezInfiniti, 2020-05-25:
Thank you... i saw that before... but as you said, just the bin file... i needed to understand how it is done... the only one i found is a .js file at
[https://github.com/NorthernMan54/homebridge-bme280/blob/master/sensor.js](url)

But this is for home bridge i guess.. how can i make it to a homekit .c file

Kristian8606, 2020-05-25:
https://github.com/Kristian8606/Temperature_sensor/blob/master/main.c 
This is my test code that works.  tuning pins 4 and 5

CodezInfiniti, 2020-05-25:
Edit:

I found one much closer. But it has i2c pins

[https://github.com/Kristian8606/Temperature_sensor](url)

CodezInfiniti, 2020-05-25:
It has i2c pins i guess... not directly connected to esp8266 pinout. Im not sure bcoz im a beginner... but i see an i2c library added initally

maximkulkin, 2020-05-25:
@CodezInfiniti you do not need to go too far, BMP280/BME280 driver is part of [ESP-OPENT-RTOS distribution](https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/bmp280). And here is [an example](https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/bmp280/bmp280_example.c) on how to use it.

CodezInfiniti, 2020-05-25:
> It has i2c pins i guess... not directly connected to esp8266 pinout. Im not sure bcoz im a beginner... but i see an i2c library added initally

Sorry...im wrong

closed 2020-06-26 by maximkulkin