maximkulkin/esp-homekit-demo #357: Gate accessory vs garage

pocketscience, 2020-06-03:
I'm using the garage example as the basis for a gate controller. It works well. However Siri doesn't know what a gate is and when I ask Siri to "close the gate" she responds with "your garage door is closing" - which is disconcerting at times as I have a commercial HomeKit garage controller installed as well. I see no mention of a gate accessory type in Apple's HomeKit documentation, so I guess this is expected behaviour..?

maximkulkin, 2020-06-04:
Are you sure it's not because you have given them such names? What name have you assigned?

pocketscience, 2020-06-04:
The gate controller is called "Gate"..

maximkulkin, 2020-06-04:
Ok. And just to clarify: are you saying that your commercial HomeKit gate opener behaves differently?

pocketscience, 2020-06-04:
Sorry if that wasn't clear. No, my commercial controller is connected to my garage - so it's fine. It's just sometimes confusing because when I ask Siri to "shut the gate" her response is "your garage is closing" - because the gate controller is registered as a garage accessory type.

Not sure there's anything actionable here other than confirming there is no "gate accessory" type available...

maximkulkin, 2020-06-04:
Not sure I understand "my commercial controller is connected to my garage - so it's fine". What is fine? What is "my garage"? If it's fine, so what is the problem? If you're referring to your own controller "not being fine", then why wouldn't you connect it to "your garage" as well, so it could also be "fine"?

> because the gate controller is registered as a garage accessory type
What is "the gate controller" ? There is not such thing as "garage" accessory type. There is a "Garage Door Opener" accessory type (at least in non-commercial specification. If you have an example that proves it wrong, I would incorporate new services definitions into this library).

pocketscience, 2020-06-04:
OK, forget about the commercial garage controller. It's attached to a different structure and tangential to the story..

When I said "garage accessory type" I meant specifically "homekit_accessory_category_garage". I'm assuming there is no "homekit_accessory_category_gate" defined (I don't see it anywhere - either in your source, or the Apple doco), which is why Siri refers to my device as a garage even though it's being used on a gate (and is called "gate").

Anyway, I think we're going in circles. I'll close this.

FWIW, Siri works very well for me...

closed 2020-06-04 by pocketscience

AramVartanyan, 2020-06-04:
@pocketscience, The best approach here is to change the name of your commercial Garage door controller in Home app and test it with Siri. If the response is the same as you mentioned above, so this issue should be addressed by the Siri developers.