maximkulkin/esp-homekit-demo #363: wifi_config: No configuration found - ESP-12F Module

kalindud, 2020-06-23:
HI,
 Im using a ESP-12F Module to try out a Sonoff_basic device.
When the device boots, it creates the AP for config. But as soon as I enter the password for the SSID,
it says: 

**!!! wifi_config: No configuration found**

and continues to scan for networks, and opens the AP again.

Why is this happening?
I tried flashing the LED.C to the same device (Which has hard-coded SSID and PASSWORD), and it works fine. 

Thank you!

pocketscience, 2020-06-25:
Sounds like it's failing to write the configuration to flash...that "No configuration found" error comes from the wifi_config_station_connect() function when it fails to retrieve the SSID/Password from the flash system.. that's why the LED example works - it's not using wifi_config to store/retrieve configuration from flash...

maximkulkin, 2020-06-25:
This is very odd. What exact device are you using? Which firmware and how have you built? Since wifi_config uses sysparams to store config and sysparams detect flash size in runtime, I suspect that sysparams properly initialized it's sectors, but then some other code (esp-homekit?) have erased them and now sysparams do not work.

kalindud, 2020-06-26:
Hi,
 Im using Firmwavre 0.9.6-dev_20150704

And the device is a ESP 12F module.

kalindud, 2020-06-26:
You are correct..... It was a problem with sys params. I was using a custom firmware for the device. and it worked after I erased the device completely before flashing the firmware.. 
Thanks for the support!!!!

closed 2020-06-26 by kalindud