maximkulkin/esp-homekit-demo #364: Memory usage

pocketscience, 2020-06-25:
I have a ESP8266 project that's regularly checking the status of a external device via HTTP. The JSON data that comes back is quite large, but manageable. I'm using malloc each time I query the device, and religiously freeing it after parsing. However I'm seeing after some hours malloc is failing (returning NULL) even though there's plenty of heap space free (determined via sdk_system_get_free_heap_size). From a cursory glance I see that there is a lot of malloc'ing going on inside esp-homekit (including many cases where return values aren't checked...), so I'm assuming the heap is getting fragmented. For example, even though there's 20K free I can't allocate 5K...

I'd rather not move to static data, or assume a worst cases scenario and malloc a large buffer at start up - but maybe I'll need to to improve the situation. At the moment if a malloc fails I don't really have any choice but to force a restart.

So my question is whether there's been any thought about moving to a more robust memory management scheme?

maximkulkin, 2020-06-25:
> So my question is whether there's been any thought about moving to a more robust memory management scheme?

Which is ..?

While there is work being done to reduce amount of allocations inside esp-homekit (check [memory-handling-improvements](https://github.com/maximkulkin/esp-homekit/tree/memory-handling-improvements) branch), there is no guarantee that memory won't ever be fragmented. I would recommend you to a) allocate buffer for your own stuff ~~while you can~~ at startup and reuse; b) try to implement data handling algorithms that works in fixed size memory.

pocketscience, 2020-06-25:
I guess I was eluding to a scheme that would reduce the impact of fragmentation. Understand it would add significant overhead. esp-homekit will fail spectacularly when the fragmentation gets to a certain level (based on the lack of checks), and might explain why other esp-homekit based systems I have (that don't allocate memory dynamically) go offline sometimes. Happy to hear memory handling is being looked at in general though. I'll move to static buffers. Cheers,

closed 2020-06-26 by pocketscience