maximkulkin/esp-homekit-demo #365: help with "make flash monitor"

renandw, 2020-06-25:
Hello. I am working on a code that allows me to control 4 relays controlled by an individual toggle, and a button to reset configuration.
But for some reason, the logs, when I use "make flash monitor", stops. But I can configure wifi, pair it to HomeKit and everything. But some bugs show up and I can't see why.

```
#include <stdio.h>
#include <stdlib.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <queue.h>
#include <string.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>

#include <button.h>
#include <toggle.h>
#include "led_status.h"
#include "config.h"

// The GPIO pin that is connected to RELAY#1 on the board.
const int relay_gpio_1 = 1;
// The GPIO pin that is connected to RELAY#2 on the board.
const int relay_gpio_2 = 3;
// The GPIO pin that is connected to RELAY#3 on the board.
const int relay_gpio_3 = 13;
// The GPIO pin that is connected to RELAY#4 on the board.
const int relay_gpio_4 = 16;

// The GPIO pin that is connected to the LED on the ESP12F.
const int led_gpio = LED_GPIO;

// The GPIO pin that is connected to the button on the Board.
#define BUTTON_PIN 0
#ifndef BUTTON_PIN
#error BUTTON_PIN is not specified
#endif

// The GPIO pin that is connected to the header on the board(external switch).
#define TOGGLE_PIN_1 5
#ifndef TOGGLE_PIN_1
#error TOGGLE_PIN_1 is not specified
#endif

#define TOGGLE_PIN_2 4
#ifndef TOGGLE_PIN_2
#error TOGGLE_PIN_2 is not specified
#endif

#define TOGGLE_PIN_3 14
#ifndef TOGGLE_PIN_3
#error TTOGGLE_PIN_3 is not specified
#endif

#define TOGGLE_PIN_4 12
#ifndef TOGGLE_PIN_4
#error TOGGLE_PIN_4 is not specified
#endif

// NOVO
// one short blink every 3 seconds
led_status_pattern_t mode_normal = LED_STATUS_PATTERN({100, -2900});
// two short blinks every 3 seconds
led_status_pattern_t mode_connecting_to_wifi = LED_STATUS_PATTERN({100, -100, 100, -2700});
// long blink, long wait
[... 333 lines cut ...]

EXTRA_COMPONENTS = \
        extras/http-parser \
        extras/dhcpserver \
  $(abspath ../../components/common/button) \
        $(abspath ../../components/esp8266-open-rtos/wifi_config) \
        $(abspath ../../components/esp8266-open-rtos/cJSON) \
        $(abspath ../../components/common/wolfssl) \
        $(abspath ../../components/common/homekit)

FLASH_SIZE ?= 8
FLASH_MODE ?= dout
FLASH_SPEED ?= 40
HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000

EXTRA_CFLAGS += -I../.. -DHOMEKIT_SHORT_APPLE_UUIDS

include $(SDK_PATH)/common.mk

monitor:
        $(FILTEROUTPUT) --port $(ESPPORT) --baud 115200 --elf $(PROGRAM_OUT)
```

is something that I am forgetting?
thanks

maximkulkin, 2020-06-25:
You're using GPIO 1 and 3 to control your relays, but they are part of UART0 interface which is used to communicate with the board over USB. The moment you do gpio_enable() on those your serial interface stops working. IMO it's OK to use GPIO3 since people rarely transmit something to MCU, but GPIO1 should not be used.

This is a pretty tight setup but doable. I've done 4-button-4-relays firmwares in the past and here is my GPIO config: buttons = [0, 2, 4, 5], relays = [12, 13, 14, 16]. I haven't done a reset button though, but you can do the following: use GPIO 0 as reset/control button and use GPIO 3 as one of button inputs.

Generally, there is a [good article on what pins can be used where](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/), which you can easily google with "esp8266 what pins to use".

renandw, 2020-06-25:
Thanks a lot @maximkulkin. I made those changes and not only the logs are fine, but the bugs are also gone. Thanks for the help

closed 2020-06-25 by renandw