maximkulkin/esp-homekit-demo #368: Sonoff_PWM Application

kalindud, 2020-07-07:
Hi, 
 I used the sonoff PWM with a AC Lightbulb using a ZVC circuit. But i used a different microcontroller IC to drive the ZVC. :D

Is it possible to modify the sonoff_pwm code to do the zvc detection and use just the IoT Device to control it?

Hope its possible... thanx

renandw, 2020-07-08:
Based on this discussion, I do not think that is possible.
[https://github.com/maximkulkin/esp-homekit-demo/issues/345](url)

kalindud, 2020-07-08:
Any information on How to use : [esp-ac-dimmer](https://github.com/maximkulkin/esp-ac-dimmer) with homekit devices?

maximkulkin, 2022-06-21:
esp-ac-dimmer is a library to control AC dimmer, the one that usually has Zero-Crossing trigger pin and a PWM pin (google "AC dimmer module"). Just connect Z-C and PWM pins to ESP, pass those to the library and specify AC frequency (needed to calculate duty cycle). You can use it e.g. to control brightness of incandescent light bulbs or dimmable LED bulbs. In HomeKit accessory firmware add BRIGHTNESS characteristic and call esp-ac-dimmer ac_dimmer_set_duty().

closed 2022-06-21 by maximkulkin