maximkulkin/esp-homekit-demo #369: error flashing

borcares, 2020-07-14:
I have this error when I intent to flash:

_Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings._

anyone can help?

thanks.

maximkulkin, 2020-07-14:
More details please. What are you trying to achieve? Which firmware? What commands do you run?

borcares, 2020-07-15:
Hi Maxim,

I've compiled the occupancy example with this configurations in the Makefile:
_FLASH_SIZE ?= 8
FLASH_MODE ?= dout
FLASH_SPEED ?= 40
HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000_

After that, I've tried to flash the wemos D1 mini and nodemcu v3 with the comand:
_esptool.py -p /dev/cu.wchusbserial1420 --baud 115200 write_flash -fs 8m -fm dout -ff 40m 0x00000 occupancy.bin_

and I've had the same warning:
_Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings_ 

Thanks in advanced.

borcares, 2020-07-17:
anyone can help?

Thanks.

maximkulkin, 2020-07-17:
Well, I guess, it is expected for bootloader to start with particular sequence and since you're flashing at bootloader sector, the program complains about it. Why did you decide to to flash it manually rather than using `make flash`? If you really know what you're doing, you need to read esptool.py source code to see how they prepare bootloader image file and be prepared to make your image compliant.

borcares, 2020-07-17:
Hi maxim, thanks for the feedback. 

I flash it manually because I’m compiling the firmware in a virtual machine and I have problems to connect the usb port with the virtual machine... and I flash it in macos. I will try to flash it into the virtual machine with the comand “make flash”. 

Thank you.

AramVartanyan, 2020-07-17:
I am using the Maxim’s Docker guide and it works perfectly under Mac OSX.
However if you would like to use esptool you should set all parameters. Also you need to copy the rboot, blank_config and your firmware bin files in one place.
rboot.bin and blank_config.bin are located here:

https://github.com/SuperHouse/esp-open-rtos/tree/master/bootloader/firmware_prebuilt

Your commands should look like this:

python esptool/esptool.py --chip esp8266 -p /dev/cu.wchusbserial1420 -b 115200 erase_flash

python esptool/esptool.py --chip esp8266 -p /dev/cu.wchusbserial1420 -b 115200 write_flash 0x00 rboot.bin 0x1000 blank_config.bin 0x2000 occupancy.bin

borcares, 2020-07-18:
Than you AramVartanyan. Finally I can flash the firmware with your comand.

closed 2020-07-18 by borcares