maximkulkin/esp-homekit-demo #370: error when executing `make -C /example/led error_flash`

Vardaan2048, 2020-07-18:
Hi Folks,

After running make -C /example/led error_flash i am getting the following errors.

make: Entering directory '/home/vardaan/Documents/Projects/Home-Kit-POC/esp-homekit-demo/examples/led'
Makefile:13: /home/user/projects/esp-open-rtos/common.mk: No such file or directory
make: *** No rule to make target '/home/user/projects/esp-open-rtos/common.mk'.  Stop.
make: Leaving directory '/home/vardaan/Documents/Projects/Home-Kit-POC/esp-homekit-demo/examples/led'

I have checked if the env variables are set, following are the results for that:

echo $PATH
/home/user/projects/esp-open-sdk/xtensa-lx106-elf/bin:/home/user/projects/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

echo $SDK_PATH
/home/user/projects/esp-open-rtos

echo $ESPPORT
/dev/ttyUSB0

echo $FLASH_SIZE
8

echo $FLASH_MODE
dout

Can anybody help with issue?
Thanks in advance

Vardaan2048, 2020-07-18:
Resolved the issue.

closed 2020-07-18 by Vardaan2048

Izac9, 2020-11-27:
where was the problem ? I have same Issue

maximkulkin, 2020-11-28:
@Izac9 I guess the solution was to follow [development guide](https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker)) and pay attention to what you type. From the original description I see a couple of issues: "/example/led" should be just "example/led" (since it is a relative directory where to look for Makefile), "error_flash" is not actually a valid target (I guess it was supposed to be "erase_flash"... or just "flash").