maximkulkin/esp-homekit-demo #375: build exampls/esp32/button

coffice12, 2020-08-06:
connect io0 buttonand gnd

click button No response

What's wrong？

renandw, 2020-08-06:
you defined GPIO 0 as your "BUTTON_PIN"?

maximkulkin, 2020-08-06:
@renandw The firmware already defines that. Not sure what is happening. @coffice12 Can you provide logs?

nick-nick-32, 2020-09-03:
Hi guys,

I’m totally noob on this platform, so I will perhaps say something wrong but, I faced the same issue and it seems there is a typo in _components > common > button > **port.c**_

For the _ESP-IDF part_, which is used for _ESP32_ if I’m not wrong, the **my_gpio_enable** function set the GPIO as **output** **instead of input**.

I gave a try to add
```c
 gpio_set_direction(button_gpio, GPIO_MODE_INPUT);
```
 directly in **app_main** of _examples > esp32 > button > main > **main.c**_ and I’m now able the see logs when button pushed.

Hopes it can help…

coffice12, 2020-09-07:
搞定！谢谢你了！thins！！

closed 2020-09-07 by coffice12