maximkulkin/esp-homekit-demo #376: HomeKit Devices shows No Response

AR-Tawil, 2020-08-12:
Hi, I'm really puzzled about the non responsive state of my devices. I noticed that my accessories becomes unresponsive immediately once my Home App is running in the background. It take around 30 seconds for the accessories to become available when I return to the home App and some even takes longer.  

I'm using an ipad as my home hub for automation and  for sharing resources. I can keep accessories responsive if I leave the home app opened on my iPad and on active state running in the foreground?!!! Also, I sometimes get a **DHCP is not healthy** network alerts but not sure why this happens? 

Can you please advise if you have encountered similar issue
[image]
 and how you managed to overcome it.

Thank you for your excellent work and appreciate your support.

AR Tawil
[image]

AR-Tawil, 2020-08-12:
This is the code I'm using:

#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

#include "mqtt_client.c"

#include <lwip/err.h>
#include <lwip/sockets.h>
#include <lwip/sys.h>
#include <lwip/netdb.h>
#include <lwip/dns.h>

/* Add extras/sntp component to makefile for this include to work */
#include <sntp.h>
#include <time.h>
#include <string.h>
#include <ota-api.h>

#define SNTP_SERVERS 	"0.pool.ntp.org", "1.pool.ntp.org", \
						"2.pool.ntp.org", "3.pool.ntp.org"

#define vTaskDelayMs(ms)	vTaskDelay((ms)/portTICK_PERIOD_MS)
#define UNUSED_ARG(x)	(void)x

time_t ts; 

int c_hash_previous = 0;

homekit_characteristic_t ota_trigger  = API_OTA_TRIGGER;
homekit_characteristic_t manufacturer = HOMEKIT_CHARACTERISTIC_(MANUFACTURER,  "IoT Smart Solutions");
homekit_characteristic_t serial       = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, "48:3F:DA:54:F6:1F");
homekit_characteristic_t model        = HOMEKIT_CHARACTERISTIC_(MODEL,         "KitcheHubLight");
homekit_characteristic_t revision     = HOMEKIT_CHARACTERISTIC_(FIRMWARE_REVISION,  "0.0.1");

char* concat(const char *s1, const char *s2, const char *s3)
{
    char *result = malloc(strlen(s1) + strlen(s2) + strlen(s3) + 1); // +1 for the null-terminator
    // in real code you would check for errors in malloc here
    strcpy(result, s1);
    strcat(result, s2);
    strcat(result, s3);
    return result;
}

void removeStringTrailingNewline(char *str) {
  if (str == NULL)
    return;
  int length = strlen(str);
  if (str[length-1] == '\n')
    str[length-1]  = '\0';
}

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
[... 109 lines cut ...]
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-009"
};

void user_init(void) {
    uart_set_baud(0, 115200);
    wifi_init();
   /*  int c_hash=ota_read_sysparam(&manufacturer.value.string_value,&serial.value.string_value,
                                      &model.value.string_value,&revision.value.string_value);
	printf("c_hash = %d\n", c_hash);
    //c_hash=1; revision.value.string_value="0.0.1"; //cheat line
    config.accessories[0]->config_number=c_hash; 
	//printf("config.accessories[000000000000000000] = %d\n", config.accessories[0]->config_number);  */
    led_init();
    homekit_server_init(&config);
    xTaskCreate(sntp_tsk, "SNTP", 1024, NULL, 1, NULL);
}

maximkulkin, 2020-08-12:
Not sure why you get "DHCP is unhealthy" messages, this might be related to your network router configuration, but that might cause issues with HomeKit. It takes some time for iOS to query accessory state after going out of sleep, so it is expected. However, accessories should not show up as "No Response", there should be a "Updating" status. Can you send debug logs from a device when it experiences a problem.

AR-Tawil, 2020-08-13:
Thank you for your reply Maxim. 

Please see below my network details and let me know if you can spot something wrong. I'm not sure why Brand and Model, Bonjour and DHCP are not showing any details. Also I noticed that my router is not able to recognise the names for my devices?!!

[image]
[image]

maximkulkin, 2020-08-13:
It says that you do not have DHCP active. That won't work. You need DHCP running.