maximkulkin/esp-homekit-demo #379: xtensa-lx106-elf-gcc: command not found

JaniniRami07, 2020-08-20:
Hey I am trying to flash the firmware on my ESP8266 but I get the follwing error when I run ```make -C examples/led_strip```:
```
make: Entering directory '/home/ramij/projects/esp-homekit-demo/examples/led_strip'
CC /home/ramij/projects/esp-open-rtos/open_esplibs/libmain/timers.c
make: xtensa-lx106-elf-gcc: Command not found
/home/ramij/projects/esp-open-rtos/open_esplibs/component.mk:15: recipe for target 'build/open_esplibs_libmain//timers.o' failed
make: *** [build/open_esplibs_libmain//timers.o] Error 127
make: Leaving directory '/home/ramij/projects/esp-homekit-demo/examples/led_strip'
```
I have installed everything from GitHub using ```-- recursive``` and still got the same error, Any idea of what might be causing it?

JaniniRami07, 2020-08-20:
I am currently working on an Ubuntu system, will docker work on Ubuntu system?

maximkulkin, 2020-08-20:
Yes it will.

JaniniRami07, 2020-08-20:
Hey, So I downloaded docker, installed python and followed the instructions and got this when trying to run ```docker run -it --rm -v "$(pwd)":/project -w /project esp-rtos make -C examples/sonoff_basic all```:
```
Unable to find image 'esp-rtos:latest' locally
```
When building the esp-rtos-dockerfile I got and error at the end which I think it causes the error above:
```
Command: docker build . -f esp-rtos-dockerfile -t esp-rtos
```
```
Error: Cloning into 'tests/unity'...
Submodule path 'tests/unity': checked out 'bbf2fe3a934f96cd00693841247a689e57a17b0d' Unable to checkout '8fe0804ceb875193a37106a2e25d07c106650a86' in submodule path 'lvgl/lv_drivers'
The command '/bin/sh -c git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos' returned a non-zero code: 1
```
Do you have an idea on what is causing this error?

xuanbka1, 2020-08-24:
> I am currently working on an Ubuntu system, will docker work on Ubuntu system?

You can download esp8266 toolchain, then you add $PATH for folder contain it,  Ubuntu will detect  _xtensa-lx106-elf-gcc_

closed 2020-09-14 by maximkulkin

anandloorthuraja, 2020-10-18:
> Hey, So I downloaded docker, installed python and followed the instructions and got this when trying to run `docker run -it --rm -v "$(pwd)":/project -w /project esp-rtos make -C [...]

Hi,
I also stuck at the same line

Could you help me how u have solved this error, Please?