maximkulkin/esp-homekit-demo #383: multi sensor pairing

jonasftw, 2020-09-02:
I cannot connect multi sensor (temperature, humidity and motion sensor) to homekit. i've tested them separately and everything worked fine. this is what i get when i try to pair multi sensor:
`>>> HomeKit: Configuring mDNS
mDNS announcement: Name=MultiSensor-ACCD
md=esp8266pv=1.0id=AC:CD:92:75:93:89c#=1s#=1ff=0sf=1ci=8 Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with D883E859-D0FF-43F1-B9D2-97532C09FCEF
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=MultiSensor-ACCD
md=esp8266pv=1.0id=AC:CD:92:75:93:89c#=1s#=1ff=0sf=0ci=8 Port=5556 TTL=4500
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with D883E859-D0FF-43F1-B9D2-97532C09FCEF
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Remove Pairing
>>> HomeKit: Removed pairing with D883E859-D0FF-43F1-B9D2-97532C09FCEF
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS`

maximkulkin, 2020-09-03:
You have mistakes in your accessory definition, it can be clearly seen that it pairs ok, then downloads accessory database and pulls back. Check or paste your accessory definition.

jonasftw, 2020-09-03:
Sorry to bother, but how do I do any of those?

maximkulkin, 2020-09-14:
You open your code in a text editor, select accessory definition (everything that goes inside HOMEKIT_ACCESSORY()), Ctrl+C/Cmd+C, open this conversation, click on reply box, Ctrl+V/Cmd+V, format it a little bit, click Comment button.

dachoeks3, 2020-10-22:
Is it possible to change mdns name? The one that shows up on your router.

maximkulkin, 2020-11-07:
mDNS name is the name if your accessory (value of NAME characteristic inside ACCESSORY_INFO service of first accessory).

closed 2020-11-07 by maximkulkin