maximkulkin/esp-homekit-demo #384: help with dimmer sample

lukasa1993, 2020-09-10:
hello i am using this extensively for some samples which are provided, i also want to use it with dimmer i have few of https://m.aliexpress.com/item/32891383747.html?trace=wwwdetail2mobilesitedetail&spm=a2g0s.9042311.0.0.300d4c4dx6emz1
this

atm using with tasmota they have template for this but its not ideal and fails many times.

so i wonder shich sample
is best to start with this

mriksman, 2020-09-10:
How is the dimming done? Does your Tasmota template use 'TuyaMCU'?

lukasa1993, 2020-09-10:
https://templates.blakadder.com/wifi_dimmer_switch.html

this is guide i followed

mriksman, 2020-09-10:
Not an easy job.

The Tasmota software sends serial commands to a second chip. 
You'd need to program that yourself. 
https://tasmota.github.io/docs/devices/PS-16-DZ-Dimmer/

It is doable though. I've done something similar for a Mitsubishi A/C - sending serial commands to the controller.

lukasa1993, 2020-09-10:
> Not an easy job.
> 
> The Tasmota software sends serial commands to a second chip.
> You'd need to program that yourself.
> https://tasmota.github.io/docs/devices/PS-16-DZ-Dimmer/
> 
> It is doable though. I've done something similar for a Mitsubishi A/C - sending serial commands to the controller.

is sample for Mitsubishi available somewhere ?

mriksman, 2020-09-10:
https://github.com/mriksman/esp32_homekit_mitsubishi
Can't offer any support though. Hopefully it'll give you some ideas.

closed 2020-11-07 by maximkulkin