maximkulkin/esp-homekit-demo #386: 220V switch question

outoforder20, 2020-09-14:
I tried to modify button lib but i could not make it.

Here some sources for 220V switch:

https://templates.blakadder.com/moes-MS-104B.html

https://tasmota.github.io/docs/Buttons-and-Switches/

https://community.home-assistant.io/t/moes-ms104-b-howto-config/172237

https://templates.blakadder.com/blitzwolf_BW-SS5_2.html

I think it is possible with right code

mriksman, 2020-09-14:
Check out https://github.com/RavenSystem/esp-homekit-devices. This uses the esp-homekit library, and is quite customisable.

maximkulkin, 2020-09-14:
@outoforder20 I agree, it should be possible with the right code

outoforder20, 2020-09-14:
https://github.com/RavenSystem/esp-homekit-devices/issues/1135

closed 2020-11-07 by maximkulkin