maximkulkin/esp-homekit-demo #389: Building with Docker: Permission denied

JohnWickham, 2020-09-20:
I’m following the instructions on the [Docker Wiki page](https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker)) using Docker 2.3.0.5 on macOS 10.15.6. Everything works great up until step 5, which fails with this error: `The command '/bin/sh -c git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos' returned a non-zero code: 1`.

Is anyone else able to follow through these instructions without issue? I’m new to using Docker so apologies if this is obvious!

shamasis, 2020-09-23:
Seems like certificate and source related issue. The following changes to the Dockerfiles worked for me:

**esp-sdk-dockerfile**
```
FROM ubuntu:18.04 as builder

RUN groupadd -g 1000 docker && useradd docker -u 1000 -g 1000 -s /bin/bash --no-create-home
RUN mkdir /build && chown docker:docker /build

RUN DEBIAN_FRONTEND="noninteractive" apt-get update && apt-get -y install tzdata

RUN apt-get update && apt-get install -y \
  make unrar-free autoconf automake libtool gcc g++ gperf \
  flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
  sed git unzip bash help2man wget bzip2 libtool-bin

RUN su docker -c " \
    git clone --recursive https://github.com/pfalcon/esp-open-sdk.git /build/esp-open-sdk ; \
    cd /build/esp-open-sdk ; \
    make STANDALONE=n ; \
"

FROM ubuntu:18.04

RUN apt-get update && apt-get install -y make python python-serial

COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
ENV PATH /opt/xtensa-lx106-elf/bin:$PATH
```

**esp-rtos-dockerfile**
```
FROM ubuntu:18.04 as builder

RUN apt-get update && apt-get install -y git

RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos

FROM esp-sdk:latest

COPY --from=builder /opt/esp-open-rtos /opt/esp-open-rtos

ENV SDK_PATH /opt/esp-open-rtos
```

Essentially, updated to Ubuntu 18.04

cc @JohnWickham @maximkulkin

maximkulkin, 2020-11-07:
Updated instructions on Wiki

closed 2020-11-07 by maximkulkin