maximkulkin/esp-homekit-demo #392: Failed to read HomeKit storage magic

alejandrogzu, 2020-09-25:
Hello Maxi,

i am having this issue and i don't know why, can you give me some help? I am using a ESP01s with 1MB.
I read in a closed issue that @AchimPieters solved with the below docker command but is not working for me.

----------------------------------------------------------------
i create the .bin file with

docker run -it --rm -v `pwd`:/project -w /project esp-rtos make -C examples/led FLASH_SIZE=8 HOMEKIT_SPI_FLASH_BASE_ADDR=0x8c000 all

-------------------------------------------------------------------
log:

rBoot4LCM v1.0.1
reformatted start_bits field: ffffffff count: 0
         ffffffff @ 1810
         ffffffff @ 1020 => count: 1
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0 at 2000, load addr 63730.
pp_task_hdl : 3ffeffe0, prio:14, stack:512
pm_task_hdl : 3ffef8d0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3ef8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Sep 23 2020 00:15:19
phy ver: 273, pp ver: 8.3

mode : sta(2c:3a:e8:17:2e:0f)
add if0
>>> HomeKit: Starting server
!!! HomeKit: Failed to read HomeKit storage magic
>>> HomeKit: Formatting HomeKit storage at 0x100000
!!! HomeKit: Failed to erase HomeKit storage
>>> HomeKit: Resetting HomeKit storage
!!! HomeKit: Failed to reset HomeKit storage
>>> HomeKit: Generated new accessory ID: 38:D9:0E:F4:A3:98
!!! HomeKit: Failed to write accessory ID to HomeKit storage
scandone
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to HomeKit storage
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Sample LED md=MyLEDpv=1.0id=38:D9:0E:F4:A3:98c#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
add 0
aid 2
cnt

maximkulkin, 2020-09-26:
Doesn't look like settings had any effect. Did you do a clean build?

alejandrogzu, 2020-09-26:
thank you Maxi, I made a clean build and now is working perfect!

closed 2020-09-27 by maximkulkin