maximkulkin/esp-homekit-demo #394: Adaptive Lighting

Kristian8606, 2020-09-29:
https://support.apple.com/en-in/guide/iphone/iph0a717a8fd/ios

Is this a new characteristic and can you add it as an example @maximKulkin

maximkulkin, 2020-09-30:
Color temperature is something that [was there for some time already](https://github.com/maximkulkin/esp-homekit/blob/master/include/homekit/characteristics.h#L88), although I'm not sure how "adaptiveness" work: should accessories tune it automatically based on clock, or it's iOS controllers who do that (by just adjusting color temperature).

Kristian8606, 2020-09-30:
`Touch and hold a lighting accessory that supports Adaptive Lighting.
Tap the Adaptive Lighting button.`
this part of the guide makes me think that the function will be supported by the firmware of the bulb.  But you will need a characteristic that will display a button to activate the function.

closed 2020-11-07 by maximkulkin

HomeKidd, 2020-11-13:
[HAP-NodeJS](https://github.com/homebridge/HAP-NodeJS/blob/d0eff7603c579f6993a715df27de2c105af6db6c/src/lib/controller/AmbientLightningController.ts#L35) now has an example for Adaptive Lighting in the beta branch 😄  With [Color Utils](https://github.com/homebridge/HAP-NodeJS/blob/beta/src/lib/util/color-utils.ts) functions too