maximkulkin/esp-homekit-demo #395: TV Support Example

joaodforce, 2020-09-30:
I making an adapter to control my LG TV via it's serial port. I got a hold of the protocol it uses, and It is possible to control anything trough it. Change volume, input. directions etc...

However Im not familiar with the ESP-Homekit SDK yet, Im studying it and the examples, however there is no example of setting the acessory to show as an actual TV on the home App. I've gotten the core methods working, I just need to bing the correct services and characteristics to make it appear right.

If someone can give me a light on the matter I would gladly make a PR to add the TV accessory as an example.

maximkulkin, 2020-09-30:
Check [this](https://github.com/maximkulkin/esp-ir-tv) repository, it's a TV accessory that controls TV through IR commands.

andrei-gavrila, 2020-09-30:
Here is an example on using the input sources - that can be freely renamed in Homekit - to set up a list of favorite channels for easy switching: https://github.com/maximkulkin/esp-ir/issues/11

More details are available in the post.

@maximkulkin BTW, a reply to that issue would be highly appreciated 😛

joaodforce, 2020-09-30:
omg! this is perfect! thanks, I'll be working with the provided examples.

pocketscience, 2020-10-09:
> I making an adapter to control my LG TV via it's serial port. I got a hold of the protocol it uses, and It is possible to control anything trough it. Change volume, input. directions etc...
>
> However [...]

@joaodforce Can you post a link to the LG protocol documentation please? Thanks.

joaodforce, 2020-10-09:
@pocketscience  Sure thing, I used the one from this site

https://support.justaddpower.com/kb/article/36-lg-rs232-control/

I also have this PDF, but I don't remember where I got it from, I think I printed from some website.

[RS-232C - LG TV Control.pdf](https://github.com/maximkulkin/esp-homekit-demo/files/5354801/RS-232C.-.LG.TV.Control.pdf)

I don't know how many LG TVs share this protocol however.

closed 2020-11-07 by maximkulkin