maximkulkin/esp-homekit-demo #396: Create homekit bridge

FFTSolution, 2020-10-21:
Hi maximkulkin,

Can I create Homekit Bridge with this library, I try create Homekit Bride on Arduino use esp8266 but i got error
**src/homekit/types.h:252:34: error: 'HOMEKIT_SERVICE_BRIDGE' undeclared here (not in a function)**

How can I solve it ?
thanks for support !

maximkulkin, 2020-10-21:
Let's first define what we are trying to achieve here.

FFTSolution, 2020-10-21:
i want to create homekit Bridge, so I want use Zigbee/Z-wave for send data from Sensor to Homebride !

maximkulkin, 2020-10-21:
Usually what happens is your bridge device exposes itself as an accessory with ID=1 and just an ACCESSORY_INFO service (and accessory category=2 (homekit_accessory_category_bridge)). All bridged devices are exposed as additional accessories, each have services corresponding to accessory type.

closed 2020-11-07 by maximkulkin